m-labs/artiq #1549: zotino.init() hangs with artiq 5

hermitdemschoenenleben, 2020-11-25:
# Bug Report

## One-Line Summary

After updating from 4 to 5, artiq_run hangs whenever zotino.init() is called.

## Issue Details

We are using kasli-hub and flashed the gateware after installing artiq 5. We tried with nix and conda versions of the packages.

### Steps to Reproduce

1. Experiment:
```python
from artiq.experiment import EnvExperiment, kernel

class BenTestSequence(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("zotino0")

    @kernel
    def run(self):
        print('start kernel')
        self.core.reset()
        self.core.break_realtime()

        self.zotino0.init()
        print('zotino initialized')
```
2.
```
[nix-shell:~/bebec]$ artiq_run --verbose repository/ben_minimal.py -v
DEBUG:artiq.compiler.import_cache:hook installed
DEBUG:artiq.compiler.import_cache:added 'repository/ben_minimal.py' to cache
DEBUG:artiq.compiler.import_cache:added '/nix/store/6avp9mpww1ddvcwz9w2m8ivc0alq92zx-python3-3.7.6-env/lib/python3.7/site-packages/artiq/coredevice/spi2.py' to cache
DEBUG:artiq.compiler.import_cache:added '/nix/store/6avp9mpww1ddvcwz9w2m8ivc0alq92zx-python3-3.7.6-env/lib/python3.7/site-packages/artiq/coredevice/ad53xx.py' to cache
DEBUG:artiq.compiler.import_cache:added '/nix/store/6avp9mpww1ddvcwz9w2m8ivc0alq92zx-python3-3.7.6-env/lib/python3.7/site-packages/artiq/coredevice/zotino.py' to cache
DEBUG:artiq.compiler.import_cache:added '/nix/store/6avp9mpww1ddvcwz9w2m8ivc0alq92zx-python3-3.7.6-env/lib/python3.7/site-packages/artiq/coredevice/ttl.py' to cache
DEBUG:artiq.coredevice.comm:connected to 192.168.0.200:1381
DEBUG:artiq.coredevice.comm_kernel:sending message: type=<Request.SystemInfo: 3>
DEBUG:artiq.coredevice.comm_kernel:receiving message: type=<Reply.SystemInfo: 2>
DEBUG:artiq.coredevice.comm_kernel:sending message: type=<Request.LoadKernel: 5>
DEBUG:artiq.coredevice.comm_kernel:receiving message: type=<Reply.LoadCompleted: 5>
DEBUG:artiq.coredevice.comm_kernel:sending message: type=<Request.RunKernel: 6>
DEBUG:artiq.coredevice.comm_kernel:running kernel
DEBUG:artiq.coredevice.comm_kernel:receiving message: type=<Reply.RPCRequest: 10>
DEBUG:artiq.coredevice.comm_kernel:rpc service: [2]<built-in function print> ['start kernel'] {} -> b'n'
start kernel
DEBUG:artiq.coredevice.comm_kernel:rpc service: 2 ['start kernel'] {} = None
DEBUG:artiq.coredevice.comm_kernel:sending message: type=<Request.RPCReply: 7>

( now nothing happens anymore )
```

### Your System (omit irrelevant parts)

* Tried on Ubuntu 18.04 and Fedora 32
*
```
[nix-shell:~/bebec]$ artiq_client --version
ARTIQ v5.7136.15bb0fa9
```
* 
```
>>> artiq_coremgmt -D 192.168.0.200 log
[     0.000009s]  INFO(runtime): ARTIQ runtime starting...
[     0.003934s]  INFO(runtime): software ident 5.7136.15bb0fa9;hub
[     0.009864s]  INFO(runtime): gateware ident 5.7136.15bb0fa9;hub
[     0.015811s]  INFO(runtime): log level set to INFO by default
[     0.021531s]  INFO(runtime): UART log level set to INFO by default
[     0.027914s]  INFO(runtime::rtio_clocking): using internal RTIO clock (by default)
[     0.304791s]  INFO(board_artiq::si5324): waiting for Si5324 lock...
[     2.852243s]  INFO(board_artiq::si5324):   ...locked
[     2.881453s]  INFO(runtime): network addresses: MAC=54-10-ec-a9-d2-7b IPv4=192.168.0.200 IPv6-LL=fe80::5610:ecff:fea9:d27b IPv6=no configured address
[     2.895732s]  INFO(runtime::mgmt): management interface active
[     2.910916s]  INFO(runtime::session): accepting network sessions
[     2.927737s]  INFO(runtime::session): running startup kernel
[     2.932253s]  INFO(runtime::session): no startup kernel found
[     2.939770s]  INFO(runtime::session): no connection, starting idle kernel
[     2.945495s]  INFO(runtime::mgmt): new connection from 192.168.0.24:40476
[     2.952565s]  INFO(runtime::session): no idle kernel found
```
* kasli-hub
*
```
>>> cat device_db.py
# most recent version should be here:
# https://github.com/m-labs/artiq/blob/master/artiq/examples/kasli_basic/device_db_hub.py
core_addr = "192.168.0.200"

device_db = {
    "core": {
        "type": "local",
        "module": "artiq.coredevice.core",
        "class": "Core",
        "arguments": {"host": core_addr, "ref_period": 1e-9}
    },
    "core_log": {
        "type": "controller",
        "host": "::1",
        "port": 1068,
        "command": "aqctl_corelog -p {port} --bind {bind} " + core_addr
    },
    "core_cache": {
        "type": "local",
        "module": "artiq.coredevice.cache",
        "class": "CoreCache"
    },
    "core_dma": {
        "type": "local",
        "module": "artiq.coredevice.dma",
        "class": "CoreDMA"
    },

    "i2c_switch0": {
        "type": "local",
        "module": "artiq.coredevice.i2c",
        "class": "PCA9548",
        "arguments": {"address": 0xe0}
    },
    "i2c_switch1": {
        "type": "local",
        "module": "artiq.coredevice.i2c",
        "class": "PCA9548",
        "arguments": {"address": 0xe2}
    },
}

device_db.update({
    "ttl" + str(i): {
        "type": "local",
        "module": "artiq.coredevice.ttl",
        "class": "TTLInOut" if i < 4 else "TTLOut",
        "arguments": {"channel": i},
    } for i in range(24)
})

device_db.update({
    "spi_sampler0_adc": {
        "type": "local",
        "module": "artiq.coredevice.spi2",
        "class": "SPIMaster",
        "arguments": {"channel": 24}
    },
    "spi_sampler0_pgia": {
        "type": "local",
        "module": "artiq.coredevice.spi2",
        "class": "SPIMaster",
        "arguments": {"channel": 25}
    },
    "spi_sampler0_cnv": {
        "type": "local",
        "module": "artiq.coredevice.ttl",
        "class": "TTLOut",
        "arguments": {"channel": 26},
    },
    "sampler0": {
        "type": "local",
        "module": "artiq.coredevice.sampler",
        "class": "Sampler",
        "arguments": {
            "spi_adc_device": "spi_sampler0_adc",
            "spi_pgia_device": "spi_sampler0_pgia",
            "cnv_device": "spi_sampler0_cnv"
        }
    }
})

for j in range(3):
    device_db.update({
        "spi_urukul{}".format(j): {
            "type": "local",
            "module": "artiq.coredevice.spi2",
            "class": "SPIMaster",
            "arguments": {"channel": 27 + 3*j}
        },
        "ttl_urukul{}_sync".format(j): {
            "type": "local",
            "module": "artiq.coredevice.ttl",
            "class": "TTLClockGen",
            "arguments": {"channel": 28 + 3*j, "acc_width": 4}
        },
        "ttl_urukul{}_io_update".format(j): {
            "type": "local",
            "module": "artiq.coredevice.ttl",
            "class": "TTLOut",
            "arguments": {"channel": 29 + 3*j}
        },
        "urukul{}_cpld".format(j): {
            "type": "local",
            "module": "artiq.coredevice.urukul",
            "class": "CPLD",
            "arguments": {
                "spi_device": "spi_urukul{}".format(j),
                "sync_device": "ttl_urukul{}_sync".format(j),
                "io_update_device": "ttl_urukul{}_io_update".format(j),
                "refclk": 100e6,
                "clk_sel": 0
            }
        }
    })

    device_db.update({
        "urukul{}_ch{}".format(j, i): {
            "type": "local",
            "module": "artiq.coredevice.ad9910",
            "class": "AD9910",
            "arguments": {
                "pll_n": 40,
                "chip_select": 4 + i,
                "cpld_device": "urukul{}_cpld".format(j)
            }
        } for i in range(4)
    })

device_db.update({
    "led0": {
        "type": "local",
        "module": "artiq.coredevice.ttl",
        "class": "TTLOut",
        "arguments": {"channel": 36}
    },
    "led1": {
        "type": "local",
        "module": "artiq.coredevice.ttl",
        "class": "TTLOut",
        "arguments": {"channel": 37}
    }
})

device_db.update({
    "spi_zotino0": {
        "type": "local",
        "module": "artiq.coredevice.spi2",
        "class": "SPIMaster",
        "arguments": {"channel": 38}
    },
    "ttl_zotino0_ldac": {
        "type": "local",
        "module": "artiq.coredevice.ttl",
        "class": "TTLOut",
        "arguments": {"channel": 39}
    },
    "ttl_zotino0_clr": {
        "type": "local",
        "module": "artiq.coredevice.ttl",
        "class": "TTLOut",
        "arguments": {"channel": 40}
    },
    "zotino0": {
        "type": "local",
        "module": "artiq.coredevice.zotino",
        "class": "Zotino",
        "arguments": {
            "spi_device": "spi_zotino0",
            "ldac_device": "ttl_zotino0_ldac",
            "clr_device": "ttl_zotino0_clr"
        }
    }
})
```

jordens, 2020-11-25:
The device database doesn't match the gateware.
Use the (matching) template from https://nixbld.m-labs.hk/job/artiq/full/device-db-kasli-hub/latest-finished containing the correct RTIO channel numbers.

hartytp, 2020-11-25:
we should really add this kind of thing to the artiq FAQ.  @jordens what do you think about tagging this as a documentation issue and leaving it open? I'm happy to spin up a doc PR at some point for this and a few other things that come up all the time.

jordens, 2020-11-25:
Sure. That works.

hermitdemschoenenleben, 2020-11-26:
thanks, the new device db worked!

In addition to adding it to the FAQ, I'd suggest a note to the "Upgrading ARTIQ" section of this document (that's where I looked for help):
https://m-labs.hk/artiq/manual/installing.html

sbourdeauducq, 2021-12-13:
The release notes already list breaking changes. We should simply make sure that those changes that require a device db modification are listed.

closed 2021-12-13 by sbourdeauducq