m-labs/artiq #1553: Queueing events blocked

hermitdemschoenenleben, 2020-11-30:
# Question

## Category: development

## Description

I'm trying to program a fast sequence using urukul and zotino with timesteps in the us regime. The total sequence will take ~1s and I'm doing some calculation inside the kernel for programing the events. In doing so, I always got Underflow Exceptions. Reading the docs, I thought I could prevent these by just putting a huge delay (`delay(10000*ms)`) in the beginning of the sequence, as this would guarantee that I'm able to program the events before the rtio counter reaches the programmed times. Unfortunately, this didn't work and I still have underflows. I tried to debug this and found out that programming lots of events in the future doesn't work: For each DDS channel, only 16 events are actually queued, before the CPU is blocked (cf. the script below) for 10 seconds until the first events are executed. Is this behavior intended? What I find very confusing in addition is that adding a second DDS statement below the first one (`self.urukul0_ch1.set_mu(0)`) results in 64 events instead of 16 programmed at once.

### Code:

```python
from artiq.experiment import EnvExperiment, kernel, ms

class BenTestSequence(EnvExperiment):
    def build(self):
        self.setattr_device("core")

        for u in range(3):
            self.setattr_device("urukul{}_cpld".format(u))
            for ch in range(4):
                self.setattr_device("urukul{}_ch{}".format(u, ch))

    def prepare(self):
        self.my_counter = 0

    @kernel
    def run(self):
        print('start kernel')
        self.core.reset()
        self.core.break_realtime()

        self.urukul0_cpld.init()
        self.urukul0_ch0.init()
        self.urukul0_ch1.init()
        self.urukul0_ch2.init()
        self.urukul0_ch3.init()

        # turn all channels on
        self.urukul0_cpld.cfg_sw(0, 1)
        self.urukul0_cpld.cfg_sw(1, 1)
        self.urukul0_cpld.cfg_sw(2, 1)
        self.urukul0_cpld.cfg_sw(3, 1)

        self.core.break_realtime()

       # huge delay in the beginning
       # idea was to advance now_mu very far to avoid underflows
        delay(10000*ms)

        for i in range(5000):
            self.urukul0_ch0.set_mu(0)

            self.my_counter = self.my_counter + 1
            print(self.my_counter)

            # long delay, just for testing
            delay(100*ms)
```

### Output

```
❯ time artiq_run repository/ben_minimal.py
start kernel
1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
         NOTE: Nothing happens here for ~10seconds
17
18
...
```

sbourdeauducq, 2020-11-30:
You are filling up the gateware RTIO buffer and then the CPU has to wait for an event to be executed before being able to submit a new one.
The buffer size can be increased at compilation time. Where did you get your Kasli system and do you know the variant?

hermitdemschoenenleben, 2020-11-30:
Thank you for the quick reply!

> The buffer size can be increased at compilation time.
Is it possible to increase this value a lot, i.e. to a few thousand? Or will this lead to problems?

> Where did you get your Kasli system and do you know the variant?
We have artiq-kasli-hub (kasli 1.1) and afaik we bought it from quartiq