m-labs/artiq #1559: Mirny: clock selection documentation

occheung, 2020-12-11:
# Bug Report

## One-Line Summary

Documentations in `mirny.py` about clock selection differs from what is mentioned in the schematics of Mirny v1.1.

## Issue Details
The comments in question:
```
    :param clk_sel: Reference clock selection.
        valid options are: 0 - internal 100MHz XO; 1 - front-panel SMA; 2 -
        internal MMCX
```
According to the [schematics for Mirny v1.1](https://github.com/sinara-hw/Mirny/releases), both 1 and 3 should refer internal MMCX, 2 should corresponds to the front-panel SMA.

Follow-up: #1530 @airwoodix

airwoodix, 2020-12-11:
Thanks. This apparently changed between v1.0 and v1.1. #1530 was developed against v1.0 and I don't have a v1.1 to test. I guess v1.0 support should be phased out anyway, since it won't be compatible with new mezzanine boards. Once tested on hardware, do you want to prepare a PR for this issue?

@jordens: related to quartiq/mirny#3.

jordens, 2020-12-11:
The price of hardware changes is software complexity or compatibility. @occheung if you want to fix this, please devise a way that does not break v1.0.

occheung, 2020-12-14:
I am only saying the comment is a bit incorrect, especially for Mirny v1.1.
Can we fix it by specifying the versions? Just like the one on Urukul?
https://github.com/m-labs/artiq/blob/4b10273a2d7047d8ad3f4bba7fbdef0282e4bcd2/artiq/coredevice/urukul.py#L133-L137

airwoodix, 2021-01-28:
@jordens would you be fine passing ``clk_sel`` as a string (e.g. ``"xo", "mmcx", "sma"``) that's interpreted according to ``hw_rev`` read in ``init``? This would make the interface much friendlier and also applies to Urukul (although for Urukul, one may need to increase the status register width to accomodate for 7 bits of ``proto_rev`` and 4 bits of ``hw_rev``). We can of course make this change backwards compatible by checking the argument's type.

jordens, 2021-01-29:
Sure. It's going to break if hw_rev isn't updated. With that and the other obvious caveats (forward, backward compatibility), you also need to look at the clocking population options especially on v1.1, different clocks for different channels, etc.
Also probably wise to choose 0b11 and not 0b10 for "mmcx" to disable the XO. Otherwise we have the same issue as on Urukul v1.0.

PR #1601 (2021-02-02): mirny: hw_rev independent, human readable clk_sel

# ARTIQ Pull Request

## Description of Changes

This patch allows for hardware-revision independent user strings instead of integers for the Mirny reference clock selection. The relevant integer is calculated from the hardware revision read from the CPLD at ``init()``. The change is fully backwards compatible. The ``clk_sel`` functionality is identical to pre-existing code, but the interface is nicer.

Tested against v1.0 and v1.1 hardware on all three global clock sources.

``hw_rev`` is not tested anymore in ``init()``. Removed correspondin [... cut]
closed 2021-02-02 by jordens