m-labs/artiq #1562: RJ45 10/100/1000 SFP failure if inserted when Kasli powers up

sbourdeauducq, 2020-12-14:
# Bug Report

## One-Line Summary

Ethernet fails if the 10/100/1000 (RGMII) is already inserted when Kasli is powered on

## Issue Details

### Steps to Reproduce

1. Insert fs.com SFP-GB-GE-T into unpowered Kasli 2.0 flashed with ARTIQ-6.
2. Power it up 
3. Try to use Ethernet.

### Expected Behavior

Ethernet works.

### Actual (undesired) Behavior

No Ethernet link. Funnily enough, the Ethernet link LED is on when the RJ45 cable is unplugged and off when it is plugged.

Removing and reinserting the SFP with the Kasli powered up works around this bug.

HarryMakes, 2020-12-14:
Just tested on a v1.1 by flashing 6.7515.3f631c41.beta and the issue is reproducible. Am looking into the issue and trying to fix it.

HarryMakes, 2020-12-23:
Should be fixed with https://github.com/m-labs/misoc/pull/105 now, at least tested on a Kasli 2.0.

closed 2020-12-23 by sbourdeauducq