m-labs/artiq #1564: artiq_browser: Application crashes when a H5 file is opened

aaditkamat, 2020-12-15:
# Bug Report

## One-Line Summary
When I launch the `artiq_browser` and open a HDF5 file generated as the result of an experiment run on ARTIQ, the application crashes.

### Steps to Reproduce
1.  Execute the `artiq_browser` command
2.  Open the HDF5 file by locating in the file system through the Explorer pane

### Expected Behavior
I expected to see a plot of the HDF5 file contents rendered by the GUI.

### Actual (undesired) Behavior
**Error message:**
```
WARNING:browser:artiq.browser.files:unable to read metadata from C:/Users/<USERNAME>/PycharmProjects/artiq-experiments/results/2020-12-14/17/000000039-TrigEnvironment.h5
Traceback (most recent call last):
  File "C:\Users\<USERNAME>\anaconda3\envs\artiq\lib\site-packages\artiq\browser\files.py", line 177, in list_current_changed
    expid = pyon.decode(f["expid"].value)
AttributeError: 'Dataset' object has no attribute 'value'
Traceback (most recent call last):
  File "C:\Users\<USERNAME>\anaconda3\envs\artiq\lib\site-packages\artiq\browser\files.py", line 199, in list_current_changed
    rd[k] = (True, v.value)
AttributeError: 'Dataset' object has no attribute 'value'
```

### Your System (omit irrelevant parts)

* Operating System: Windows 10 Pro Version 2004 
* ARTIQ version: ARTIQ v6.7504.6ceb3f30.beta
* If using Conda, output of `conda list` (please submit as a file attachment, as this tends to be long)
[conda-list.txt](https://github.com/m-labs/artiq/files/5694955/conda-list.txt)

dnadlinger, 2020-12-15:
Without remembering whether this is Python or Julia, didn't the interface change to `[()]` instead of `.value`? Should be a simple patch.

aaditkamat, 2020-12-16:
Yeah, it's a Python Dataset object created using the `h5py`  library.

dnadlinger, 2020-12-16:
Sure; I reworded the message to clarify – I was referring to [this](https://docs.h5py.org/en/stable/whatsnew/2.1.html#dataset-value-property-is-now-deprecated), but wasn't 100% sure whether this was indeed in h5py or a similar library. artiq_browser needs a fix to use the new API.

sbourdeauducq, 2020-12-16:
A regression test for this would be good, we can just import the relevant functions from the browser and run them without involving the GUI.

sbourdeauducq, 2020-12-16:
> artiq_browser needs a fix to use the new API.

And we should probably detect which API to use and make ``artiq_browser`` compatible with both. Especially because most people use the trashy conda software installer and its ridiculously buggy dependency handler, and sometimes overwrite parts of the packages with pip and other messy installers - the combination of which makes is difficult to control which h5py version gets imported.

dnadlinger, 2020-12-16:
h5py 2.0 (with the working `[()]` syntax) was released in 2014, though; surely, that would be okay to assume by now? `.value` has been deprecated for a long time, and only now actually removed.

sbourdeauducq, 2020-12-16:
I don't see h5py < 2.0 available in popular conda channels, so yes it should be ok...

dnadlinger, 2020-12-16:
@aaditkamat: Do you want to submit a PR? Should be a simple matter of find/replace.

aaditkamat, 2020-12-17:
> @aaditkamat: Do you want to submit a PR? Should be a simple matter of find/replace.

I have done a replacement in the `files.py` as suggested but I'm not sure about which tests I need to update.

sbourdeauducq, 2020-12-17:
See ``artiq/test``.
This code isn't covered by tests atm (which is why this wasn't flagged by CI); I was suggesting adding a new test.

aaditkamat, 2020-12-17:
> See `artiq/test`.
> This code isn't covered by tests atm (which is why this wasn't flagged by CI); I was suggesting adding a new test.

You mentioned a regression test invoking the relevant functions from the browser but how about adding a test case under [`test_h5types.py`](https://github.com/m-labs/artiq/blob/master/artiq/test/test_h5types.py)

sbourdeauducq, 2020-12-17:
That would be fine, rename ``test_h5types`` to ``test_hdf5`` since it would be testing more than data types then.

closed 2020-12-17 by sbourdeauducq