m-labs/artiq #1570: DRTIO link does not come up between Kasli 2.0

cjbe, 2020-12-19:
# Bug Report

A DRTIO link cannot be established between two Kasli 2.0's.
The master SFP1 and slave SFP0 are connected by fibre transceivers. The master SFP link light turns on, but the satellite link light does not. No DRTIO link is established.

We have tried different transceivers, and different SFP ports on the master. In all cases the behaviour is the same. The SFP0 of the satellite works with an ethernet SFP when the satellite is flashed as a master. This weakly suggests that this is a software / configuration problem rather than a hardware problem.

### Actual (undesired) Behavior

Master core log:
```
...
[  1016.096514s]  INFO(runtime::rtio_mgt::drtio): [LINK#1] link RX became up, pinging
[  1036.252677s] ERROR(runtime::rtio_mgt::drtio): [LINK#1] ping failed
[  1036.458121s]  INFO(runtime::rtio_mgt::drtio): [LINK#1] link RX became up, pinging
[  1056.614272s] ERROR(runtime::rtio_mgt::drtio): [LINK#1] ping failed
[  1056.819602s]  INFO(runtime::rtio_mgt::drtio): [LINK#1] link RX became up, pinging  
```

Satellite core log:
```
Booting from flash...
Starting firmware.
[0.000005s]  INFO(satman): ARTIQ satellite manager starting...
[     0.005612s]  INFO(satman): software ident 6.unknown.beta;lasersatellite   
[     0.012327s]  INFO(satman): gateware ident 6.unknown.beta;lasersatellite    
[     0.365784s]  INFO(board_artiq::si5324): waiting for Si5324 lock...        
[     2.298061s]  INFO(board_artiq::si5324):   ...locked    
```

### Your System (omit irrelevant parts)

* ARTIQ version: based on 3e3883302001eaee5c46ff33726ef16b25537fb7
* Hardware involved: 2x Kasli 2.0

cjbe, 2020-12-19:
This is possibly related to #1562

hartytp, 2020-12-19:
Are the transceiver fault indicators still high?

cjbe, 2020-12-19:
Unknown - I am working remotely, and don't have a readout for that. I understand that at one point @sotirova observed this, but I am not sure with what level of confidence.

hartytp, 2020-12-19:
It was definitely the case in her setup previously. I’m happy to sketch out some pseudo code to add readout of them if it helps?

cjbe, 2020-12-19:
Sure - that would be useful (especially if you are happy to close the loop with @sotirova when implementing)

jordens, 2020-12-20:
Drive TX_DISABLE.
https://github.com/m-labs/artiq/pull/1508

hartytp, 2020-12-20:
> Drive TX_DISABLE

I'm aware of that issue but don't believe that's the cause of what we're seeing. @sotirova and Fabian measured the voltages on all outputs of the two I2C expanders on the master + satellite. TX_DISABLE is low here, so it's not an issue with the particular transceivers we're using -- although I do agree that this should generally be driven low in the firmware; that's been on by to do list for a while now.

What's interesting is that we do see a TX_FAULT on the master side of the DRTIO link. My understanding (from talking to @sbourdeauducq is that this indicates an internal fault with the transceiver) so I assumed it was just a faulty transceiver. However, @cjbe reports having tried multiple transceivers without success, so that's maybe less likely. It feels worth reporting the state of TX_DISABLE when we get failed pings. Again, that's a PR that's been on my to do list for some time.

[image]

[image]

[image]
[image]
[image]

> or use a DAC.

Can you elaborate on that, I don't follow. Which DAC do you mean?

hartytp, 2020-12-20:
doh, nope, I've just made a transcription error in the above. Master SFP1 TX_DISABLE is IC25 pin 2 which should be at 0V but is at 2V0. So the fix is just to extend Pawel's patch to drive it low.

jordens, 2020-12-20:
DAC as in direct attach cable.

commit 43be383c (2020-12-22): kasli v2.0: drive TX_DISABLE low on all SFPs (fixes #1570)

This was the same problem as #1508 but on SFP1..3
closed 2020-12-22 by sbourdeauducq