m-labs/artiq #1575: nix artiq-full.novatech409b

jbqubit, 2021-01-07:
I added ```artiq-full.novatech409b``` in my nix environment. I get the following error. 

```
$ nix-shell my-artiq-env.nix
these derivations will be built:
  /nix/store/mvkkig2wv9pi1r8ihicdplwl4a40yf31-python3-3.7.6-env.drv
building '/nix/store/mvkkig2wv9pi1r8ihicdplwl4a40yf31-python3-3.7.6-env.drv'...
collision between `/nix/store/1wjbsk3sq38qwvm45l3hpx459113dds8-python3.7-lda-1.1/lib/python3.7/site-packages/test/__pycache__/__init__.cpython-37.pyc' and `/nix/store/52yqfn454x8ib50qqpjdd7yfaghfyz6i-python3.7-novatech409b-1.1/lib/python3.7/site-packages/test/__pycache__/__init__.cpython-37.pyc'
builder for '/nix/store/mvkkig2wv9pi1r8ihicdplwl4a40yf31-python3-3.7.6-env.drv' failed with exit code 25
error: build of '/nix/store/mvkkig2wv9pi1r8ihicdplwl4a40yf31-python3-3.7.6-env.drv' failed
```

I see similar errors for ```artiq-full.korad_ka3005p``` and ```lda```.

sbourdeauducq, 2021-01-07:
Please post the exact nix-shell file (and exact other commands if applicable) that can be used to reproduce the problem.

drewrisinger, 2021-01-07:
Looking at this bug, I'd assume that the issue is that the ``test`` directory of ``python3Packages.lda`` and ``python3Packages.novatech409b`` is being copied to the outputs (``$out``), causing a conflict. This issue is resolved on newer versions of nixpkgs with https://github.com/NixOS/nixpkgs/blob/master/pkgs/development/interpreters/python/hooks/python-remove-tests-dir-hook.sh, but it apparently didn't make it to ``nixos-20.03`` (actually, this hook still wouldn't resolve the issue b/c the test directory is called "test" and not "tests" as the hook expects).

This is probably an issue with the MLabs ``nix-scripts`` repo, and can be resolved with adding something like ``postFixup = "rm -rf $out/${python.sitePackages}/test"`` to those packages.

jbqubit, 2021-01-07:
```
cat my-artiq-env.nix 
let
  # pkgs contains the NixOS package collection. ARTIQ depends on some of them, and
  # you may want some additional packages from there.
  pkgs = import <nixpkgs> {};
  artiq-full = import <artiq-full> { inherit pkgs; };
in
  pkgs.mkShell {
    buildInputs = [
      (pkgs.python3.withPackages(ps: [
        # List desired Python packages here.

        # You probably want these two.
        artiq-full.artiq
        artiq-full.artiq-comtools

        # You need a board support package if and only if you intend to flash
        # a board (those packages contain only board firmware).
        # The lines below are only examples, you need to select appropriate
        # packages for your boards.
        #artiq-full.artiq-board-kc705-nist_clock
        #artiq-full.artiq-board-kasli-wipm
        #ps.paramiko  # needed if and only if flashing boards remotely (artiq_flash -H)
	artiq-full.wand
        artiq-full.lda
        artiq-full.korad_ka3005p
        artiq-full.novatech409b
        artiq-full.thorlabs_tcube
        # The NixOS package collection contains many other packages that you may find
        # interesting for your research. Here are some examples:
        #ps.pandas
        #ps.numpy
        #ps.scipy
        #ps.numba
        #(ps.matplotlib.override { enableQt = true; })
        #ps.bokeh
        #ps.cirq
        #ps.qiskit
        #ps.qutip
      ]))

      # List desired non-Python packages here
      #artiq-full.openocd  # needed if and only if flashing boards
      # Other potentially interesting packages from the NixOS package collection:
      #pkgs.gtkwave
      #pkgs.spyder
      #pkgs.R
      #pkgs.julia
    ];
  }

```

sbourdeauducq, 2021-01-15:
fixed

closed 2021-01-15 by sbourdeauducq

jbqubit, 2021-01-16:
Confirmed. Thanks!