m-labs/artiq #1576: unit test for inclusion of optional nix packages

jbqubit, 2021-01-07:
A test is needed to confirm that optional nix packages can be added to nix shell environment without error. See https://github.com/m-labs/artiq/issues/1575. Examples of such packages include the following.

        ps.paramiko
	artiq-full.wand
        artiq-full.lda
        artiq-full.korad_ka3005p
        artiq-full.novatech409b
        artiq-full.thorlabs_tcube
        ps.pandas
        ps.numpy
       ps.scipy
        ps.numba
        (ps.matplotlib.override { enableQt = true; })
        ps.bokeh
        ps.cirq
        ps.qiskit
        ps.qutip
      artiq-full.openocd  
      pkgs.gtkwave
      pkgs.spyder
      pkgs.R
      pkgs.julia

drewrisinger, 2021-01-07:
Because this is a Nix problem, I don't think an ARTIQ unit test is the appropriate place. If it was anywhere, it should probably be a build artifact of https://git.m-labs.hk/M-Labs/nix-scripts (i.e. show up on the MLabs Hydra server https://nixbld.m-labs.hk/jobset/artiq/full). 

However, this is tricky, because the ``mkShell`` nix command **explicitly does not support building** (i.e. can't be called w/ ``nix build``, only ``nix-shell``). I can't think of an easy workaround off the top of my head, though I admit I could be missing something obvious (maybe a VM test like in https://github.com/NixOS/nixpkgs/tree/master/nixos/tests?).

References:
* https://fzakaria.com/2020/08/11/caching-your-nix-shell.html
* https://nixos.wiki/wiki/Caching_nix_shell_build_inputs

jbqubit, 2021-01-07:
Ideally an automated process discovers problems adding ARTIQ-related packages to a nix environment. Better than end users discovering the problem. Problems should be flagged whether the source is an ARTIQ-level nix script error, a package error or an error with nix itself.

sbourdeauducq, 2021-01-07:
mkShell is just a special and very simple derivation: https://github.com/NixOS/nixpkgs/blob/master/pkgs/build-support/mkshell/default.nix
It could be replaced with ``runCommand`` that does ``touch $out`` for testing. A VM test would be rather heavy, and unnecessary unless there are some special test commands that could be run there that do need a VM.

sbourdeauducq, 2021-01-15:
done

closed 2021-01-15 by sbourdeauducq