m-labs/artiq #1580: Phaser does not work with current Artiq master

cjbe, 2021-01-14:
# Bug Report

## One-Line Summary

Phaser does not work with the current Artiq master - the init() fails, and bypassing this no RF is produced.

## Issue Details

I am using the kasli_generic tooling to build a trivial firmware/gateware with just 1x Phaser. I have loaded the Phaser with gateware built from master, and am using the example.py experiment from the phaser repository.

The experiment errors-out at the DUC+Oscillator test - the values of data_i and data_q are 0.
Bypassing this test, no RF is produced.

Using the firmware/gateware posted [here](https://github.com/quartiq/phaser/issues/5#issuecomment-709409539), the self tests pass, and RF is generated.

I have also tested this with the v0.4 phaser gateware, with no change in behaviour (still no RF after bypassing tests)

### Your System (omit irrelevant parts)

Using a Kasli v1.1
Building gateware with Vivado 2020.1

cjbe, 2021-01-14:
@jordens can you reproduce this? (AFAIK you are the only person outside Oxford with a Phaser)

jordens, 2021-01-14:
I can't right now. We did test this just before christmas and it worked fine.
You may want to check what changed since then. AFAICT it's only the fixing of the number of Phaser RTIO channels in the ddb generator and the FIR replacement.

commit 261870bd (2021-01-19): phaser: fix oscillator rtio address for even base addresses

close #1580
closed 2021-01-19 by jordens