m-labs/artiq #1581: Urukul cfg_sw() of one channel resets the CPLD switch of another channel on the same board

steine01, 2021-01-14:
# Bug Report

## One-Line Summary

When switching on an Urukul channel TTL with `cfg_sw()`, the other Urukul channels on the same board are set to a value corresponding to their `sw` attribute value (either from the device_db entry or from other instances set before).

## Issue Details

### Steps to Reproduce

1. I switch on `urukul0_ch0` and `urukul0_ch1` via `self.urukul0_ch0.cfg_sw(1)` and `self.urukul0_ch1.cfg_sw(1)` in a separate experiment.
2. I run the following experiment:

```
class UrukulTTLA(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("urukul0_ch0")
        
    @kernel
    def run(self):
        self.core.break_realtime()
        self.urukul0_ch0.cfg_sw(1)
```

### Expected Behavior

Both `urukul0_ch0` and  `urukul0_ch1` stay on.

### Actual (undesired) Behavior

`urukul0_ch0` stays on, but `urukul0_ch1` switches off. 

### Further investigation

If I execute the following experiment,  `urukul0_ch1` is switched on and stays on, even if I execute `UrukulTTLA` now.

```
class UrukulTTLB(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("urukul0_ch1")
        
    @kernel
    def run(self):
        self.core.break_realtime()
        self.urukul0_ch1.sw.on()
```

### Your System (omit irrelevant parts)

* Operating System: Windows 10
* ARTIQ version: ARTIQ v6.7480.a97b4633.beta
* Hardware involved: 
```
device_db["urukul0_cpld"] = {
    "type": "local",
    "module": "artiq.coredevice.urukul",
    "class": "CPLD",
    "arguments": {
        "spi_device": "spi_urukul0",
        "sync_device": None,
        "io_update_device": "ttl_urukul0_io_update",
        "refclk": 100000000.0,
        "rf_sw": 0b1100,
        "clk_sel": 1
    }
}
```

jordens, 2021-01-14:
This is documented: `rf_sw – Initial CPLD RF switch register setting (default: 0x0). Knowledge of this state is not transferred between experiments.` It's a consequence of the configuration register/status register design. This can be changed by changing the CPLD firmware and API.

steine01, 2021-01-14:
I previously understood this to be the configuration of the SW TTL line, that is mentioned in the documentation of `cfg_sw` of the AD9912, which can be accessed through the `sw` attribute. But yes, now that I read it again, I see what you mean. But why is this not mentioned in the documentation of cfg_sw` of the AD9912?

jordens, 2021-01-14:
It seems better placed in the CPLD documentation. Not every bit of information is copied to every possible location where one might look for it. You can argue that there should be a reference to the CPLD docs and I'd be happy to accept a PR. That PR should also synchronize the AD9910 and AD9912 docs.