m-labs/artiq #1594: Applets not working with nixpkgs 20.09: no running event loop

lriesebos, 2021-01-28:
# Bug Report

## One-Line Summary

Applets do not work and the dashboard raises a similar error at startup.
@sbourdeauducq , I expect this to be caused by the recent change to nixpkgs 20.09.

## Issue Details

### Steps to Reproduce

1. start `artiq_master`
2. start `artiq_dashboard`, error will occur but dashboard starts. see the reduced log below.

```
[nix-shell:~/projects/kc705]$ artiq_dashboard   
...
INFO:dashboard:root:ARTIQ dashboard 5.7148.cd3e9567 connected to ::1
ERROR:dashboard:artiq.gui.state:Uncaught exception attempting to save state
Traceback (most recent call last):
  File "/nix/store/4scv5krhvyi29a7002g6mwz18cx5k4xb-python3-3.8.5-env/lib/python3.8/site-packages/artiq/gui/state.py", line 77, in _do
    await asyncio.sleep(self.autosave_period)
  File "/nix/store/v72cj06nk69cynckz2s12rhar25k1h7v-python3-3.8.5/lib/python3.8/asyncio/tasks.py", line 637, in sleep
    loop = events.get_running_loop()
RuntimeError: no running event loop
```

3. open any applet, an other error will occur but dashboard remains open. see the reduced log below.

```
...
WARNING:dashboard:artiq.gui.applets:Applet Plot XY failed to start
Traceback (most recent call last):
  File "/nix/store/4scv5krhvyi29a7002g6mwz18cx5k4xb-python3-3.8.5-env/lib/python3.8/site-packages/artiq/gui/applets.py", line 127, in start_process
    await self.ipc.create_subprocess(
  File "/nix/store/4scv5krhvyi29a7002g6mwz18cx5k4xb-python3-3.8.5-env/lib/python3.8/site-packages/sipyco/pipe_ipc.py", line 52, in create_subprocess
    self.process = await asyncio.create_subprocess_exec(
  File "/nix/store/v72cj06nk69cynckz2s12rhar25k1h7v-python3-3.8.5/lib/python3.8/asyncio/subprocess.py", line 236, in create_subprocess_exec
    transport, protocol = await loop.subprocess_exec(
  File "/nix/store/v72cj06nk69cynckz2s12rhar25k1h7v-python3-3.8.5/lib/python3.8/asyncio/base_events.py", line 1630, in subprocess_exec
    transport = await self._make_subprocess_transport(
  File "/nix/store/v72cj06nk69cynckz2s12rhar25k1h7v-python3-3.8.5/lib/python3.8/asyncio/unix_events.py", line 202, in _make_subprocess_transport
    watcher.add_child_handler(transp.get_pid(),
  File "/nix/store/v72cj06nk69cynckz2s12rhar25k1h7v-python3-3.8.5/lib/python3.8/asyncio/unix_events.py", line 1293, in add_child_handler
    loop = events.get_running_loop()
RuntimeError: no running event loop
```

### Expected Behavior

A working dashboard with applets.

### Actual (undesired) Behavior

1. Applets do not work anymore
2. Dashboard raises an error at startup, but general dashboard functions seem to be ok (we can run an experiment from the dashboard)

### Your System (omit irrelevant parts)

* Operating System: Ubuntu 20.04.1
* ARTIQ version: ARTIQ v5.7148.cd3e9567
* latest nix channels:
```
[nix-shell:~/projects/kc705]$ nix-channel --list
artiq-full https://nixbld.m-labs.hk/channel/custom/artiq/full/artiq-full
nixpkgs https://nixos.org/channels/nixos-20.09
```

sbourdeauducq, 2021-01-28:
I've reverted to 20.03, but I don't remember seeing this issue on 20.09...

lriesebos, 2021-01-28:
That's a bummer. At least 2 machines here have these errors, both running on very similar systems. Also with a minimal Nix environment and ARTIQ project I can reproduce it.

Please let me know how I can provide more information to debug this issue.

sbourdeauducq, 2021-01-28:
Could it be the sipyco update?

hartytp, 2021-01-28:
That looks like the error we got using quamash on py 3.8 due to the changes in asyncio. It should have been fixed by qasync

lriesebos, 2021-01-28:
Sipyco does not seem to make a difference on the 20.09 channel. No problems occur when switching back to 20.03 (also with sipyco 1.2).

I have no experience with quamash, but if that is the problem and qasync is not an option, maybe 20.09 with Python 3.7 is an alternative.

sbourdeauducq, 2021-01-28:
Ok, shall keep release-5 on nixpkgs 20.03 then.

closed 2021-01-28 by sbourdeauducq

drewrisinger, 2022-04-25:
@lriesebos thanks for the bug report, this bit me today. Helped having a working solution already ready.