m-labs/artiq #1599: LLVM IR call lacks parameter attributes

pca006132, 2021-01-29:
# Bug Report

## One-Line Summary
Call instruction does not contain `byval` parameter attributes for functions requiring it.

Example: call to `dma_record_start` would be lowered into the following call:
```
call void @dma_record_start({ i8*, i32 }* %.9), !dbg !32
```
but the declaration of the function is:
```
declare void @dma_record_start({ i8*, i32 }* byval)
```

## Issue Details

### Steps to Reproduce

1. Compile the following code, and dump the IR
```python
from artiq.experiment import *

class DMAPulses(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("core_dma")

    @kernel
    def record(self):
        with self.core_dma.record("pulse"):
            pass

    @kernel
    def run(self):
        self.record()
```

### Expected Behavior

The call contains a `byval` attribute for the argument.

### Actual (undesired) Behavior

* The attribute is missing. This causes misoptimization in newer LLVM versions, and it may cause other memory corruption issues as it is UB.

### Your System (omit irrelevant parts)

* ARTIQ version: current master

dnadlinger, 2021-02-25:
https://github.com/numba/llvmlite/pull/702 – we don't need for this to enter upstream, of course, but I wanted to see whether there is any immediate feedback.

closed 2021-11-27 by dnadlinger

sbourdeauducq, 2021-11-27:
@dnadlinger thanks!

commit 63b5727a (2021-11-27): compiler: Also emit byval argument attributes at call sites

See previous commit.

GitHub: Fixes #1599.