m-labs/artiq #1603: command line call of applet fails in nix shell

jbqubit, 2021-02-09:
## One-Line Summary
Command line call of applet fails in nix shell. 

I can run applet fine from within artiq_session. 

### System 

* Operating System:
* ARTIQ version: ARTIQ v6.unknown.beta (pull from 2/8/2021)

```
$ nix-shell --pure -I artiqSrc=artiq nix-scripts/artiq-fast/shell-dev.nix
```

```
$ python plot_xy.py flopping_f_brightness --x flopping_f_frequency --fit flopping_f_fit
qt.qpa.plugin: Could not find the Qt platform plugin "xcb" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Aborted (core dumped)
```

sbourdeauducq, 2021-02-09:
Try adding ``wrapQtAppsHook`` to your ``python3.withPackages`` or apply it somehow to your entire shell environment.
https://nixos.wiki/wiki/Qt#qt.qpa.plugin:_Could_not_find_the_Qt_platform_plugin_.22xcb.22_in_.22.22

jbqubit, 2021-02-10:
I don't know nix well enough to implement your suggested change. Can you reproduce my error on your system?

jbqubit, 2021-02-12:
```
$ cat nix-scripts/artiq-fast/shell-dev.nix
{ pkgs ? import <nixpkgs> {}}:

let
  artiqpkgs = import ./default.nix { inherit pkgs; };
  vivado = import ./vivado.nix { inherit pkgs; };
in
  pkgs.mkShell {
    buildInputs = [
      vivado
      pkgs.gnumake
      (pkgs.python3.withPackages(ps: (with ps; [ jinja2 jsonschema numpy paramiko ]) ++ (with artiqpkgs; [ wrapQtAppsHook migen microscope misoc jesd204b migen-axi artiq ])))
      artiqpkgs.cargo
      artiqpkgs.rustc
      artiqpkgs.binutils-or1k
      artiqpkgs.binutils-arm
      artiqpkgs.llvm-or1k
      artiqpkgs.openocd
    ];
    TARGET_AR="or1k-linux-ar";
  }

```

```
$ nix-shell --pure -I artiqSrc=artiq nix-scripts/artiq-fast/shell-dev.nix
error: undefined variable 'wrapQtAppsHook' at /home/britton/gitlab/brittonlab/nix-scripts/artiq-fast/shell-dev.nix:11:83
(use '--show-trace' to show detailed location information)
```

jbqubit, 2021-02-12:
Ping @rgresia

sbourdeauducq, 2021-02-13:
Remove ``wrapQtAppsHook`` (turns out it cannot be used easily to wrap an entire nix-shell) and just add this to your shell.nix:
```
    shellHook=
     ''
     export QT_PLUGIN_PATH=${pkgs.qt5.qtbase}/${pkgs.qt5.qtbase.dev.qtPluginPrefix}
     export QML2_IMPORT_PATH=${pkgs.qt5.qtbase}/${pkgs.qt5.qtbase.dev.qtQmlPrefix}
     '';
```

closed 2021-02-13 by sbourdeauducq

jbqubit, 2021-02-13:
Thanks! Confirmed that the following works. Please add this to your .nix recipes so this works for everyone.  :) 

```
$ cat nix-scripts/artiq-fast/shell-dev.nix
{ pkgs ? import <nixpkgs> {}}:

let
  artiqpkgs = import ./default.nix { inherit pkgs; };
  vivado = import ./vivado.nix { inherit pkgs; };
in
  pkgs.mkShell {
    buildInputs = [
      vivado
      pkgs.gnumake
      (pkgs.python3.withPackages(ps: (with ps; [ jinja2 jsonschema numpy paramiko ]) ++ (with artiqpkgs; [ migen microscope misoc jesd204b migen-axi artiq ])))
      artiqpkgs.cargo
      artiqpkgs.rustc
      artiqpkgs.binutils-or1k
      artiqpkgs.binutils-arm
      artiqpkgs.llvm-or1k
      artiqpkgs.openocd
    ];
    TARGET_AR="or1k-linux-ar";
     shellHook=
     ''
     export QT_PLUGIN_PATH=${pkgs.qt5.qtbase}/${pkgs.qt5.qtbase.dev.qtPluginPrefix}
     export QML2_IMPORT_PATH=${pkgs.qt5.qtbase}/${pkgs.qt5.qtbase.dev.qtQmlPrefix}
     '';
  }

$ nix-shell --pure -I artiqSrc=artiq nix-scripts/artiq-fast/shell-dev.nix

```