m-labs/artiq #1607: documentation: HDFView > 3.x

jbqubit, 2021-02-09:
ARTIQ manual [mentions](https://m-labs.hk/artiq/manual/getting_started_mgmt.html?highlight=hdf5) existence of HDFView but doesn't recommend version number. Older versions of HDFView show ARTIQ hdf5 archives as [empty](https://github.com/m-labs/artiq/issues/1340). Please consider updating manual to specify HDFView >=3.x and/or adding supported version to nix shell. This is an easy pitfall that one of my students just encountered.

dhslichter, 2021-02-10:
This seems like something that is HDFView's fault.  Why would a student install an old version of HDFView instead of the latest version?

sbourdeauducq, 2021-02-10:
nixpkgs has an old version (2.14). Do you happen to know if hdf_java needs to be updated as well?

sbourdeauducq, 2021-02-10:
I think that's not a "documentation" issue, this should be just updated in nixpkgs.

jbqubit, 2021-02-10:
> Why would a student install an old version of HDFView instead of the latest version?

Depending on the Ubuntu distro version different versions of HDFView are installed as noted in the [old Issue](https://github.com/m-labs/artiq/issues/1340). It's not uncommon for Ubuntu to be full release numbers behind on tools like HDFView. 

> I think that's not a "documentation" issue, this should be just updated in nixpkgs.

Agreed that it will be easier to maintain (vs documentation) if the right version is included the default ARTIQ nixpkgs.

jbqubit, 2021-02-10:
> Do you happen to know if hdf_java needs to be updated as well?

I've never explicitly used hdf_java. I don't know if it is a dependency of HDFView.

astro, 2021-02-15:
> nixpkgs has an old version (2.14). Do you happen to know if hdf_java needs to be updated as well?

> I think that's not a "documentation" issue, this should be just updated in nixpkgs.

There has never been a version other than hdfview-2.14 and hdf_java-3.3.2 in nixpkgs. They are already the [latest official releases](https://support.hdfgroup.org/products/java/index.html) unfortunately.

sbourdeauducq, 2021-02-18:
OK, it's merged now and should be updated in nixpkgs 21.03 (in a couple months). How important is it to support older nixpkgs?

jbqubit, 2021-02-18:
It's OK if it takes a month or two for this fix to hit artiq nix.

astro, 2021-02-18:
It is [going to be May 2021](https://github.com/NixOS/rfcs/pull/80).

sbourdeauducq, 2021-07-05:
Confirmed hdfview-3.1.2 in nixpkgs 21.05 (used for ARTIQ 6 and 7).

closed 2021-07-05 by sbourdeauducq

jbqubit, 2021-07-05:
Thanks @astro @sbourdeauducq !