m-labs/artiq #1609: kasli build failures

astro, 2021-02-11:
# Bug Report

## One-Line Summary

Not sure if this has been noticed but a few build failures are currently hidden in Hydra's evaluation error tab: https://nixbld.m-labs.hk/jobset/artiq/full-beta#tabs-errors

## Log

```
hydra-eval-jobs returned exit code 1:
cannot build on 'localhost': don't know how to open Nix store 'localhost'
building '/nix/store/4jvh4wkqgdgcxa2sgvm3arf08w7zndgq-artiq-board-kasli-bonn1master-6.7571.af411de6.beta-src.drv'...
building
building
DIO (EEM0) starting at RTIO channel 0x000000
DIO (EEM0) starting at RTIO channel 0x000000
DIO (EEM1) starting at RTIO channel 0x000008
DIO (EEM1) starting at RTIO channel 0x000008
DIO (EEM2) starting at RTIO channel 0x000010
DIO (EEM2) starting at RTIO channel 0x000010
DIO (EEM3) starting at RTIO channel 0x000018
DIO (EEM3) starting at RTIO channel 0x000018
Traceback (most recent call last):
Traceback (most recent call last):
  File "/nix/store/0yhk4sk4x9s9hsrf3p1skbfy1pwd1rbf-python3-3.8.5/lib/python3.8/runpy.py", line 194, in _run_module_as_main
  File "/nix/store/0yhk4sk4x9s9hsrf3p1skbfy1pwd1rbf-python3-3.8.5/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
    return _run_code(code, main_globals, None,
  File "/nix/store/0yhk4sk4x9s9hsrf3p1skbfy1pwd1rbf-python3-3.8.5/lib/python3.8/runpy.py", line 87, in _run_code
  File "/nix/store/0yhk4sk4x9s9hsrf3p1skbfy1pwd1rbf-python3-3.8.5/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
    exec(code, run_globals)
  File "/nix/store/ifmgh6nggxylfbcggrha43njjzkfpk32-python3-3.8.5-env/lib/python3.8/site-packages/artiq/gateware/targets/kasli_generic.py", line 169, in <module>
  File "/nix/store/ifmgh6nggxylfbcggrha43njjzkfpk32-python3-3.8.5-env/lib/python3.8/site-packages/artiq/gateware/targets/kasli_generic.py", line 169, in <module>
    main()
    main()
  File "/nix/store/ifmgh6nggxylfbcggrha43njjzkfpk32-python3-3.8.5-env/lib/python3.8/site-packages/artiq/gateware/targets/kasli_generic.py", line 163, in main
  File "/nix/store/ifmgh6nggxylfbcggrha43njjzkfpk32-python3-3.8.5-env/lib/python3.8/site-packages/artiq/gateware/targets/kasli_generic.py", line 163, in main
    soc = cls(description, gateware_identifier_str=args.gateware_identifier_str, **soc_kasli_argdict(args))
    soc = cls(description, gateware_identifier_str=args.gateware_identifier_str, **soc_kasli_argdict(args))
  File "/nix/store/ifmgh6nggxylfbcggrha43njjzkfpk32-python3-3.8.5-env/lib/python3.8/site-packages/artiq/gateware/targets/kasli_generic.py", line 86, in __init__
  File "/nix/store/ifmgh6nggxylfbcggrha43njjzkfpk32-python3-3.8.5-env/lib/python3.8/site-packages/artiq/gateware/targets/kasli_generic.py", line 86, in __init__
    eem_7series.add_peripherals(self, description["peripherals"])
    eem_7series.add_peripherals(self, description["peripherals"])
  File "/nix/store/ifmgh6nggxylfbcggrha43njjzkfpk32-python3-3.8.5-env/lib/python3.8/site-packages/artiq/gateware/eem_7series.py", line 129, in add_peripherals
  File "/nix/store/ifmgh6nggxylfbcggrha43njjzkfpk32-python3-3.8.5-env/lib/python3.8/site-packages/artiq/gateware/eem_7series.py", line 129, in add_peripherals
    peripheral_processors[peripheral["type"]](module, peripheral, **kwargs)
    peripheral_processors[peripheral["type"]](module, peripheral, **kwargs)
  File "/nix/store/ifmgh6nggxylfbcggrha43njjzkfpk32-python3-3.8.5-env/lib/python3.8/site-packages/artiq/gateware/eem_7series.py", line 66, in peripheral_suservo
  File "/nix/store/ifmgh6nggxylfbcggrha43njjzkfpk32-python3-3.8.5-env/lib/python3.8/site-packages/artiq/gateware/eem_7series.py", line 66, in peripheral_suservo
    eem.SUServo.add_std(module,
    eem.SUServo.add_std(module,
  File "/nix/store/ifmgh6nggxylfbcggrha43njjzkfpk32-python3-3.8.5-env/lib/python3.8/site-packages/artiq/gateware/eem.py", line 503, in add_std
  File "/nix/store/ifmgh6nggxylfbcggrha43njjzkfpk32-python3-3.8.5-env/lib/python3.8/site-packages/artiq/gateware/eem.py", line 503, in add_std
    cls.add_extension(
    cls.add_extension(
  File "/nix/store/ifmgh6nggxylfbcggrha43njjzkfpk32-python3-3.8.5-env/lib/python3.8/site-packages/artiq/gateware/eem.py", line 30, in add_extension
  File "/nix/store/ifmgh6nggxylfbcggrha43njjzkfpk32-python3-3.8.5-env/lib/python3.8/site-packages/artiq/gateware/eem.py", line 30, in add_extension
    target.platform.add_extension(cls.io(eem, *args, **kwargs))
    target.platform.add_extension(cls.io(eem, *args, **kwargs))
  File "/nix/store/ifmgh6nggxylfbcggrha43njjzkfpk32-python3-3.8.5-env/lib/python3.8/site-packages/artiq/gateware/eem.py", line 478, in io
  File "/nix/store/ifmgh6nggxylfbcggrha43njjzkfpk32-python3-3.8.5-env/lib/python3.8/site-packages/artiq/gateware/eem.py", line 478, in io
    + Urukul.io_qspi(*eems[2:4], iostandard=iostandard))
    + Urukul.io_qspi(*eems[2:4], iostandard=iostandard))
  File "/nix/store/ifmgh6nggxylfbcggrha43njjzkfpk32-python3-3.8.5-env/lib/python3.8/site-packages/artiq/gateware/eem.py", line 138, in io_qspi
  File "/nix/store/ifmgh6nggxylfbcggrha43njjzkfpk32-python3-3.8.5-env/lib/python3.8/site-packages/artiq/gateware/eem.py", line 138, in io_qspi
    for i, j, iostandard, sig in ttls:
    for i, j, iostandard, sig in ttls:
ValueError: not enough values to unpack (expected 4, got 3)
ValueError: not enough values to unpack (expected 4, got 3)

builder for '/nix/store/4jvh4wkqgdgcxa2sgvm3arf08w7zndgq-artiq-board-kasli-bonn1master-6.7571.af411de6.beta-src.drv' failed with exit code 1
cannot build derivation '/nix/store/f0khlsr2h250zblss42pb3a2iywhk2kq-artiq-board-kasli-bonn1master-6.7571.af411de6.beta-gateware.nar.base64.drv': 1 dependencies couldn't be built
error: "\u001b[31;1merror:\u001b[0m\u001b[34;1m --- Error --- hydra-eval-jobs\u001b[0m\nbuild of '/nix/store/f0khlsr2h250zblss42pb3a2iywhk2kq-artiq-board-kasli-bonn1master-6.7571.af411de6.beta-gateware.nar.base64.drv' failed"
error: --- EndOfFile --- hydra-eval-jobs
unexpected EOF reading a line
```

sbourdeauducq, 2021-02-13:
Fixed this issue, now there is an obscure error that is difficult to debug due to https://git.m-labs.hk/M-Labs/nix-scripts/issues/36

sbourdeauducq, 2021-02-13:
Maybe it's OOM

sbourdeauducq, 2021-02-13:
Yes, it's OOM. Hopefully will get better after https://git.m-labs.hk/M-Labs/nix-scripts/pulls/41

closed 2021-02-13 by sbourdeauducq