m-labs/artiq #1613: Kasli 2.0 does not work with tutorial code led.py

fanmingyu212, 2021-02-22:
# Question

`artiq_run led.py` hangs with Kasli 2.0.

## Category: installation/setup

## Description

Hello. I am trying to set up a new artiq system with Kasli 2.0. I followed the documentation to install an artiq environment in anaconda (Windows 10, artiq package version 7.7579.8deb269b.beta). I am able to find the IP address of the Kasli (local address "192.128.1.75"), and the address responds to the `ping` command only when the Kasli is physically connected to the ethernet.

I met some problems when trying to run a kernel on the Kasli 2.0 though. Following the [manual](https://m-labs.hk/artiq/manual-beta/getting_started_core.html), I copied `device_db.py` from `examples/kasli` within the artiq package, to a directory containing `led.py` from the above tutorial link, and changed `core_addr` to "192.168.1.75" (the Kasli IP address). The directory that the manual mentions (`examples/master`) does not seem to exist in the current version of artiq. Then I changed `led` to `led0` in the `led.py` file, as `led` device is not defined in `device_db.py`. After that, I run `artiq_run led.py` in a command line, but it hangs there without finishing or outputting anything. If I control+C to stop the program, it outputs:
```
forrtl: error (200): program aborting due to control-C event
Image              PC                Routine            Line        Source
libifcoremd.dll    00007FFFAAF83B58  Unknown               Unknown  Unknown
KERNELBASE.dll     00007FFFF36C6293  Unknown               Unknown  Unknown
KERNEL32.DLL       00007FFFF5447BD4  Unknown               Unknown  Unknown
ntdll.dll          00007FFFF626CE51  Unknown               Unknown  Unknown
```

A related question is that is the `device_db.py` file in `examples/kasli` up to date? A tutorial on how to modify the `device_db.py` file for custom hardware configuration seems helpful to new users, and the current "device database" section does not seem to contain enough information (e.g., do EEM connectors correspond to channel numbers?). Please let me know if I missed it somewhere in the manual, otherwise I would be happy to contribute to this section once I learned how to use it.

sbourdeauducq, 2021-02-22:
> A related question is that is the `device_db.py` file in `examples/kasli` up to date?

Generally this file won't be suitable for your system. There should have been a USB stick shipped with your system, which contains the correct device_db for you.

In the case of Kasli 2.0, there are no built-in LEDs controlled by RTIO. You should use one of the TTL outputs instead.

fanmingyu212, 2021-02-22:
@sbourdeauducq thanks for the quick reply. I will try the file in the USB stick tomorrow and post the result here.

fanmingyu212, 2021-02-23:
Okay after some trials and errors I finally got it working with the help of @sbourdeauducq. Some of the details I had to do are listed here as they might be helpful for future users.

The main problem is that the gateware/firmware (I am not sure which one is the correct term here) is not compatible with the artiq software version we use. To solve this problem, the board package corresponding to the software version number has to be used (We use the M-Labs version of sinara hardware so there is a conda repository that we can get the board package from, but not sure how to do it if you use generic sinara hardware). The gateware/firmware flashing process is mostly documented on the manual, but there are several subtle points as raised and solved in #1502 in case you use a Windows computer to do it.

closed 2021-02-23 by fanmingyu212