m-labs/artiq #1615: `with parallel` conflict between Sampler and Zotino

bks-nist, 2021-02-23:
# Bug Report

## One-Line Summary

Inside a `with parallel` block, the order of access to Sampler and Zotino seems to unexpectedly matter with `nist3` hardware.

## Issue Details

### Steps to Reproduce

```python
from artiq.experiment import *

class SamplerZotinoConflict(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("sampler0")
        self.setattr_device("zotino0")

        self.setup()

    @kernel
    def setup(self):
        self.core.reset()
        self.core.break_realtime()
        self.sampler0.init()
        self.zotino0.init()

    @kernel
    def run_ok(self):
        data = [0] * 8
        self.core.break_realtime()
        with parallel:
            self.zotino0.write_dac_mu(0, 0)
            self.sampler0.sample_mu(data)

    @kernel
    def run_fail(self):
        data = [0] * 8
        self.core.break_realtime()
        with parallel:
            self.sampler0.sample_mu(data)
            self.zotino0.write_dac_mu(0, 0)

    run = run_fail
```

Save and submit with `artiq_run`.

### Expected Behavior

`run = run_ok` and `run = run_fail` should have identical behavior. The only difference is the order of the `self.sampler0.sample_mu()` and `self.zotino0.write_dac_mu()` calls inside a `with parallel` block where the order of the statements should be irrelevant.

### Actual (undesired) Behavior

`run = run_fail` gives an `RTIOUnderflow`, while `run = run_ok` does not:

```
(artiq-6) c:\Users\tsc1\Documents>artiq_run "Python Scripts\sampler_zotino_conflict.py"
Core Device Traceback (most recent call last):
  File "Python Scripts/sampler_zotino_conflict.py", line 34, in artiq_run_sampler_zotino_conflict.SamplerZotinoConflict.run_fail(..., ...) (RA=+0x6e0)
    self.zotino0.write_dac_mu(0, 0)
  File "<artiq>\coredevice/ad53xx.py", line 276, in ... artiq.coredevice.ad53xx.AD53xx.write_dac_mu<artiq.coredevice.zotino.Zotino>(...) (inlined)
    self.bus.write(
  File "<artiq>\coredevice/spi2.py", line 225, in ... artiq.coredevice.spi2.SPIMaster.write<artiq.coredevice.spi2.SPIMaster>(...) (inlined)
    rtio_output((self.channel << 8) | SPI_DATA_ADDR, data)
  File "ksupport/rtio.rs", line 67, column 14, in (Rust function)
    <unknown>
                 ^
artiq.coredevice.exceptions.RTIOUnderflow(0): RTIO underflow at 371427075399872 mu, channel 35, slack -12424 mu
Traceback (most recent call last):
  File "C:\Users\tsc1\Anaconda3\envs\artiq-6\Scripts\artiq_run-script.py", line 9, in <module>
    sys.exit(main())
  File "C:\Users\tsc1\Anaconda3\envs\artiq-6\lib\site-packages\artiq\frontend\artiq_run.py", line 225, in main
    return run(with_file=True)
  File "C:\Users\tsc1\Anaconda3\envs\artiq-6\lib\site-packages\artiq\frontend\artiq_run.py", line 211, in run
    raise exn
  File "C:\Users\tsc1\Anaconda3\envs\artiq-6\lib\site-packages\artiq\frontend\artiq_run.py", line 204, in run
    exp_inst.run()
  File "C:\Users\tsc1\Anaconda3\envs\artiq-6\lib\site-packages\artiq\language\core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "C:\Users\tsc1\Anaconda3\envs\artiq-6\lib\site-packages\artiq\coredevice\core.py", line 137, in run
    self.comm.serve(embedding_map, symbolizer, demangler)
  File "C:\Users\tsc1\Anaconda3\envs\artiq-6\lib\site-packages\artiq\coredevice\comm_kernel.py", line 644, in serve
    self._serve_exception(embedding_map, symbolizer, demangler)
  File "C:\Users\tsc1\Anaconda3\envs\artiq-6\lib\site-packages\artiq\coredevice\comm_kernel.py", line 636, in _serve_exception
    raise python_exn
artiq.coredevice.exceptions.RTIOUnderflow: RTIO underflow at 371427075399872 mu, channel 35, slack -12424 mu
```

### Your System (omit irrelevant parts)

* Operating System: Windows 10 (64-bit)
* ARTIQ version: 
```
(artiq-6) c:\Users\tsc1\Documents>artiq_client --version
ARTIQ v6.7582.c2248278
```
* Version of the gateware and runtime loaded in the core device:
```
(artiq-6) c:\Users\tsc1\Documents>artiq_coremgmt -D 192.168.0.10 log
[     0.000009s]  INFO(runtime): ARTIQ runtime starting...
[     0.003931s]  INFO(runtime): software ident 6.7582.c2248278;nist3
[     0.010036s]  INFO(runtime): gateware ident 6.7582.c2248278;nist3
```
* If using Conda, output of `conda list` : [conda-list.txt](https://github.com/m-labs/artiq/files/6032316/conda-list.txt)
* Hardware involved: Kasli, Sampler, Zotino

lriesebos, 2021-02-24:
I am not one of the main developers of ARTIQ, but I think I can explain it.

The short answer is: the `sample_mu()` function reads values back, and therefore leaves no slack. 

Now let's look a bit deeper into the decomposition of the parallel statement.

```python
@kernel
def run_fail():
    with parallel:
        self.sampler0.sample_mu(data)
        self.zotino0.write_dac_mu(0, 0)

@kernel
def run_fail_unrolled():
    t_start = now_mu()
    self.sampler0.sample_mu(data)  # Leaves no slack
    at_mu(t_start)  # Rollback timeline to mimic parallel behavior
    self.zotino0.write_dac_mu(0, 0)  # Underflow here because there is no slack
```

Function `run_fail()` is you failing function. If we unroll that parallel statement, then the machine will execute something like `run_fail_unrolled()` (the CPU is still sequential). Now due to the order and with the knowledge that `sample_mu()` leaves no slack, it is clear that the operation on the zotino will cause an underflow.

If you now switch the order of the two function calls, like you do in `run_ok()`, then the operation on the zotino will be called first. This function does leave slack for the `sample_mu()` function to work correctly.

Hope this helps!

sbourdeauducq, 2021-02-24:
@lriesebos thanks, that's correct.

closed 2021-02-24 by sbourdeauducq

bks-nist, 2021-02-24:
Thank you for the explanation, that makes sense. I would argue that it is still a bug, _semantically_[1], but the explanation makes it clear that **WONTFIX** is a reasonable answer since it would take a much smarter compiler to do anything about it.

[1]: Because if `with parallel` means anything semantically (rather than being defined by its current implementation), the order of operations within such a block should be unimportant.

dnadlinger, 2021-02-24:
`with parallel` has a clearly defined semantic meaning, just not the one you apparently expected.

bks-nist, 2021-03-02:
Here's a documentation patch which corrects the description of `with parallel`'s semantics, since the current behavior is stated above to be correct, but does not match the documentation: https://gist.github.com/bks-nist/69f379e63e7e058adecaf6a58b31ce92

jordens, 2021-03-02:
If you cut down on the bashing and submit a PR with accurate statements, we'd be happy to discuss and update the docs.

* It's not an "illusion of parallelism".
* The events are truly executed in parallel.
* There is actual parallel computing logic (lots of it) involved.
* It's not the gating or the input events that stall the CPU, but the CPU actively waiting for them. You could just as well generate events within parallel. Just retrieve them afterwards.

bks-nist, 2021-03-02:
The should I file a bug against Sampler that it doesn't have an asynchronous fetch function? The patch is certainly open for discussion and I am always willing to learn more; however, the fact that order-of-statements *matters* inside a `with parallel` block is a substantial gotcha that should be documented. (Perhaps "illiusion" is the wrong word, "asynchronous coprocessor" would be more strictly correct.) The events are executed in parallel, but submitted serially. This fact occasionally is visible to the programmer, and does not currently generate diagnostics, so documentation is the best that can be done.

I am happy to try again on wording if documentation patches are welcome. I will submit a PR, as requested.

hartytp, 2021-03-02:
@bks-nist Let's take a step back here for a minute.

Firstly, it's not helpful to jump to labelling things as bugs because they don't meet your expectations. Secondly -- while I think that everyone agrees that there are subtleties around the `with parallel` language feature which trip newcomers up regularly and should be documented better -- I don't think that the PR you filed does much to improve on the situation. If we strip out the things like "illusion" which are pretty unhelpful, much of what remains (e.g. how sequence errors behave) is not specific to parallel blocks, but rather just how ARTIQ works.

AFAICT the broader issue here is that the real-time guarantee provided by ARTIQ is a bit subtle and takes a while to get one's head around. Maybe more worked examples in the docs would help here.

hartytp, 2021-03-02:
IMO the real point here is that
```python
with parallel:
   foo
   bar
   baz
```
compiles down to something like
```python
t0 = now_mu()
foo
at(t0)
bar
at(t0)
baz
at(t0)
```

Stating that explicitly would probably do more than anything to clear this up.

The issues around sequence errors etc are encountered in many contexts and not part of the parallel design

hartytp, 2021-03-02:
The fact that the ordering of statements can affect whether or not a program runs without errors is a general feature of ARTIQ that's a common gotcha again not related to parallel blocks. The real-time guarantee ARTIQ is that everything will be deterministic or that there will be an error. So, for example, the function `foo` may work really reliably when run in isolation, but if `bar` is run before it then it may deterministically fail for a number of reasons (e.g. sequence errors since the previous function filled up all FIFOS). This kind of behaviour is pretty fundamental to the approach taken by ARTIQ. See e.g. delta-flow for a sketch of a different way of doing things.

hartytp, 2021-03-02:
> The should I file a bug against Sampler that it doesn't have an asynchronous fetch function

Finally from me on this: I don't understand the question here. What would an asynchronous fetch function be? How would that work? Is this really a Samper feature request (it's definitely more a feature request than a "bug"), or a feature request for an invasive change to the RTIO interface.

hartytp, 2021-03-02:
> The should I file a bug against Sampler that it doesn't have an asynchronous fetch function

Or, maybe what you're after here essentially boils down to adding extra API to Sampler to have separate `convert` and `read` functions? i.e. splitting `acquire` into two pieces? If that's what you're after you can certainly do it (i.e update the DAC and pulse CNV simultaneously and then read the sample from the RTIO FIFO later on)...

https://github.com/m-labs/artiq/blob/b8f4c6b9bb8ce4f2d7dc50e87690ad336841357c/artiq/coredevice/sampler.py#L122-L131

bks-nist, 2021-03-02:
I agree that Sampler would be a feature request, and possibly a rather invasive one. I apologize for calling that a bug; I used "bug" instead of the more generic "issue" (using GitHub's terminology) that I meant. One possible high-level implementation would be to asynchronously capture the SPI read into a large shift register (with RTIO events for the SPI clock cycles, but those are output events and can be scheduled in parallel with other things), then have a separate call which is _only_ the (blocking) RTIO read of that shift register. That would allow (e.g.) SPI reads from multiple Sampler cards to execute in parallel rather than be fully serialized as they are now. Or, for the example

hartytp, 2021-03-02:
hmm...that sounds not a million miles away from my previous post, but I'm still not fully sure I understand what you're after. e.g. how many events you want to capture, can you do this within the current framework of the RTIO input FIFOs, etc.

I feel like there is maybe a bit of a misunderstanding here about how some of the basic ARTIQ RTIO infrastructure works (microprocessor writing to FIFOS, etc). Much as we all like to moan about ARTIQ documentation (and there is plenty of room for improvement) it's actually about the best documented open source project I've ever seen. There is quite a bit of discussion about how this all works and it might be worth a refresher on that.