m-labs/artiq #1621: Conda: no module named llvmlite_artiq

lriesebos, 2021-02-28:
# Bug Report

## One-Line Summary

It seems the `llvmlite_artiq` package is not found when using conda. Maybe related to #1469 .

## Issue Details

### Steps to Reproduce

1. `conda create -n env artiq`
2. In python run `import artiq.coredevice.core`

### Expected Behavior

No import error

### Actual (undesired) Behavior

A full log can be seen in [this CI job](https://gitlab.com/duke-artiq/dax/-/jobs/1061968145), though not everyone can access those logs at this moment (will be accessible by everyone later this week).

The verdict can be seen below:

```
(artiq6) leon@ubuntu-2004:~$ python
Python 3.8.8 (default, Feb 24 2021, 21:46:12) 
[GCC 7.3.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import artiq
>>> artiq.__version__
'6.7582.c2248278'
>>> import artiq.coredevice.core
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/leon/miniconda3/envs/artiq6/lib/python3.8/site-packages/artiq/coredevice/core.py", line 12, in <module>
    from artiq.compiler.module import Module
  File "/home/leon/miniconda3/envs/artiq6/lib/python3.8/site-packages/artiq/compiler/module.py", line 13, in <module>
    from . import prelude, types, transforms, analyses, validators
  File "/home/leon/miniconda3/envs/artiq6/lib/python3.8/site-packages/artiq/compiler/transforms/__init__.py", line 12, in <module>
    from .llvm_ir_generator import LLVMIRGenerator
  File "/home/leon/miniconda3/envs/artiq6/lib/python3.8/site-packages/artiq/compiler/transforms/llvm_ir_generator.py", line 9, in <module>
    from llvmlite_artiq import ir as ll, binding as llvm
ModuleNotFoundError: No module named 'llvmlite_artiq'
>>> 
(artiq6) leon@ubuntu-2004:~$ conda list | grep llvm
llvmlite-artiq            0.23.0.dev                    0    https://conda.m-labs.hk/artiq
(artiq6) leon@ubuntu-2004:~$ 
```

### Your System (omit irrelevant parts)

* Operating System: Ubuntu 20.04
* ARTIQ version: 6.7582.c2248278
* `conda list`: see [this CI log](https://gitlab.com/duke-artiq/dax/-/jobs/1061968145)

sbourdeauducq, 2021-03-01:
Not related to https://github.com/m-labs/artiq/issues/1469 - the latter is Windoze-specific and fixed by workarounds such as "open the graphical IDE at least once after installing VS so that it silently finishes the setup, otherwise the command line linker will use the wrong DLLs" :roll_eyes: 

``llvmlite_artiq`` is a binary package that tends to be specific to a Python version. In your case you have 3.8, whereas we are building with 3.7. Try to use the same Anaconda version that we build the packages with, and downgrade Python to 3.7:
https://git.m-labs.hk/M-Labs/nix-scripts/src/branch/master/artiq-fast/conda/builder-env.nix#L10

lriesebos, 2021-03-01:
I see, I guess Conda on Linux is too optimistic by settling on Python 3.8. We only use that combo for CI.

It already works by just limiting the Python version of the environment with `python<3.8`, tnx for the quick reply!

closed 2021-03-01 by lriesebos

sbourdeauducq, 2021-03-01:
> It already works by just limiting the Python version of the environment with `python<3.8`, tnx for the quick reply!

I could add this to the ARTIQ dependencies if that helps.

lriesebos, 2021-03-01:
I guess that would make sense, then the Conda environment should resolve correct instantly. I'll leave that decision up to you.

sbourdeauducq, 2021-03-01:
> I guess that would make sense, then the Conda environment should resolve correct instantly.

Yes, in theory, but there are plenty of bugs in the conda dependency solver.

>  I guess Conda on Linux is too optimistic by settling on Python 3.8.

I think it does that on all platforms. On our CI, conda does not have access to the network, so it does not see the newer Python version at all. Before we kept having this sort of issue, which is beyond our control and with poor reproducibility with no way to completely rollback.

lriesebos, 2021-03-01:
Well, I think it then definitely makes sense to add a `'python<3.8'` dependency to ARTIQ on Conda. Then at least the Python version it resolves is correct.

sbourdeauducq, 2021-03-01:
Done, https://git.m-labs.hk/M-Labs/nix-scripts/commit/b8a803a64fd6523d81a1f774969773fd5a84ec7d
The channel will update automatically.