m-labs/artiq #1626: Unexpected behavior for the compiler inferring the type of numpy array on artiq 5

KaifengC, 2021-03-09:
# Bug Report

## One-Line Summary

The compiler fails to infer the type of NumPy array when dealing with the following code on artiq 6. But it works on artiq 5.

## Issue Details

### Steps to Reproduce

Run the following code on artiq 6.7526 and artiq 5.7114 will see different result.
It works on artiq 5 but not on artiq 6.

```python
from artiq.experiment import *
import numpy as np

class Bug(EnvExperiment):
    """ Bug
    """
    def build(self):
        self.core = self.get_device('core')
        self.points = None
        self.n = 0 # this must be set to 0 for reproducing this bug.

    def prepare(self):
        plist = [0 for _ in range(self.n)]
        self.n = np.int32(len(plist))
        self.points = np.array(plist, dtype=np.float64)

    @kernel
    def run(self):
        scans = self.points # bug is here
        for pi in scans:
            self.do_measure(pi)  # it is necessary to call this function both here
        self.do_measure(0.0)     # and here to reproduce this bug

    @kernel
    def do_measure(self, point):
        result = 0
```

On artiq 6, it threw out an error:

```bash
root:While compiling <repository>\test_exp.py
<repository>\test_exp.py:23:9-23:14: error: the type of this expression cannot be fully inferred
        scans = self.points # bug is here
        ^^^^^                            
<repository>\test_exp.py:23:9-23:14: note: the expression has type 'a
        scans = self.points # bug is here
        ^^^^^                            
```

### Expected Behavior

This code should have the same behavior under artiq 5 and artiq 6.

### Actual (undesired) Behavior

See the error message.

### Your System (omit irrelevant parts)

* Operating System: Windows 7
* ARTIQ version: artiq 6.7526 and artiq 5.7714
* Version of the gateware and runtime loaded in the core device: did not use core device
* If using Conda, output of `conda list` (please submit as a file attachment, as this tends to be long)
* Hardware involved: did not use hardware

dnadlinger, 2021-03-10:
Thanks for the bug report. The issue here is that the array is empty, so the only way the type can be inferred is through the loop variable being used to call the same function that is later called with `0.0`. There are probably two separate issues here: 1) We should embed host NumPy values (self.points) in a way where the type is specified separately from the values, and 2), type inference should (?) work in that case still. Fixing either would make this code compile.

KaifengC, 2021-03-11:
Thanks for checking. I get your point. I think it not surprising that people sometimes need to scan over an array without knowing if it is empty. Following your suggestion, I found a workaround that could make this work. But, I think we should support empty arrays, though.

```python
from artiq.experiment import *
import numpy as np

class Bug(EnvExperiment):
    """ Workaround for #1626
    """
    def build(self):
        self.core = self.get_device('core')

    def run(self):
        n_points = 0
        self.points = np.array([_ for _ in range(n_points)], dtype=np.float64)
        if len(self.points) != 0: # workaround: we need to ensure the array is not empty 
            self._run_on_core()   # before entering core device
        self._clean_up()

    @kernel
    def _run_on_core(self):
        for pi in self.points: # if self.point is empty, artiq 6 would throw an error
            self.do_measure(pi)

    @kernel
    def do_measure(self, point):
        print('scanning:', point)

    @kernel
    def _clean_up(self):
        self.do_measure(0.0)
```

dnadlinger, 2021-03-11:
Oh, yes, it should definitely be supported! This was just a note to other developers/my future self on what is going on here.

commit 5ba22c11 (2021-03-14): compiler: Change type inference rules for empty array() calls

array([...]), the constructor for NumPy arrays, currently has the
status of some weird kind of macro in ARTIQ Python, as it needs
to determine the number of dimensions in the resulting array
type, which is a fixed type parameter on which inference cannot
be performed.

This leads to an ambiguity for empty lists, which could contain
elements of arbitrary type, including other lists (which would
add to the number of dimensions).

Previously, I had chosen to make array([]) to be of completely
indeterminate type for this reason. Howeve [... cut]
commit c1413a99 (2021-03-15): compiler: Change type inference rules for empty array() calls

array([...]), the constructor for NumPy arrays, currently has the
status of some weird kind of macro in ARTIQ Python, as it needs
to determine the number of dimensions in the resulting array
type, which is a fixed type parameter on which inference cannot
be performed.

This leads to an ambiguity for empty lists, which could contain
elements of arbitrary type, including other lists (which would
add to the number of dimensions).

Previously, I had chosen to make array([]) to be of completely
indeterminate type for this reason. Howeve [... cut]
closed 2021-03-14 by dnadlinger