m-labs/artiq #1627: Compilation error when dealing with numpy index on artiq 6

KaifengC, 2021-03-09:
# Bug Report

## One-Line Summary

Artiq 6 will throw a compilation error when indexing the numpy array in the following way. 
But artiq 5 works well with the same code.

## Issue Details

### Steps to Reproduce

Run the following code on artiq 6.7526 and 5.7114.
It works well on artiq 5 but not on 6.

```python
from artiq.experiment import *
import numpy as np

class AssertBug(EnvExperiment):
    def build(self):
        self.core = self.get_device('core')

    def prepare(self):
        self.n = 2
        self._data = np.zeros((self.n, self.n))

    @kernel # must run on core device
    def run(self):
        length = self.n              # This line is necessary, not sure if it is also a bug since   
        #print(self._data[:self.n])  # this line won't work for both artiq 5 and 6
        print(self._data[:length])   # The bug is here

```

### Expected Behavior

Artiq 6 should have the same behavior as artiq 5 for this code.

### Actual (undesired) Behavior

I get an error:
```bash

root:Terminating with exception (AssertionError)
Traceback (most recent call last):
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\artiq\master\worker_impl.py", line 306, in main
    exp_inst.run()
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\artiq\language\core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\artiq\coredevice\core.py", line 129, in run
    self.compile(function, args, kwargs, set_result)
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\artiq\coredevice\core.py", line 109, in compile
    attribute_writeback=attribute_writeback)
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\artiq\compiler\module.py", line 77, in __init__
    self.artiq_ir = artiq_ir_generator.visit(src.typedtree)
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\artiq\compiler\transforms\artiq_ir_generator.py", line 178, in visit
    return self._visit_one(obj)
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\pythonparser\algorithm.py", line 32, in _visit_one
    return getattr(self, visit_attr)(node)
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\artiq\compiler\transforms\artiq_ir_generator.py", line 206, in visit_ModuleT
    self.generic_visit(node)
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\pythonparser\algorithm.py", line 27, in generic_visit
    self.visit(getattr(node, field_name))
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\artiq\compiler\transforms\artiq_ir_generator.py", line 174, in visit
    self.visit(elt)
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\artiq\compiler\transforms\artiq_ir_generator.py", line 178, in visit
    return self._visit_one(obj)
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\pythonparser\algorithm.py", line 32, in _visit_one
    return getattr(self, visit_attr)(node)
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\artiq\compiler\transforms\artiq_ir_generator.py", line 354, in visit_QuotedFunctionDefT
    self.visit_function(node, is_internal=True, is_quoted=True, flags=node.flags)
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\artiq\compiler\transforms\artiq_ir_generator.py", line 316, in visit_function
    result = self.visit(node.body)
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\artiq\compiler\transforms\artiq_ir_generator.py", line 174, in visit
    self.visit(elt)
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\artiq\compiler\transforms\artiq_ir_generator.py", line 178, in visit
    return self._visit_one(obj)
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\pythonparser\algorithm.py", line 32, in _visit_one
    return getattr(self, visit_attr)(node)
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\artiq\compiler\transforms\artiq_ir_generator.py", line 370, in visit_Expr
    result = self.visit(node.value)
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\artiq\compiler\transforms\artiq_ir_generator.py", line 178, in visit
    return self._visit_one(obj)
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\pythonparser\algorithm.py", line 32, in _visit_one
    return getattr(self, visit_attr)(node)
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\artiq\compiler\transforms\artiq_ir_generator.py", line 2419, in visit_CallT
    args     = [self.visit(arg_node) for arg_node in node.args]
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\artiq\compiler\transforms\artiq_ir_generator.py", line 2419, in <listcomp>
    args     = [self.visit(arg_node) for arg_node in node.args]
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\artiq\compiler\transforms\artiq_ir_generator.py", line 178, in visit
    return self._visit_one(obj)
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\pythonparser\algorithm.py", line 32, in _visit_one
    return getattr(self, visit_attr)(node)
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\artiq\compiler\transforms\artiq_ir_generator.py", line 1204, in visit_SubscriptT
    self.append(ir.SetElem(other_value, other_index, elem))
  File "D:\ProgramData\Anaconda3\envs\artiq601\lib\site-packages\artiq\compiler\ir.py", line 847, in __init__
    assert builtins.get_iterable_elt(lst.type) == value.type.find()
AssertionError
```

### Your System (omit irrelevant parts)

* Operating System: Window 7.
* ARTIQ version: artiq 6.7526 and 5.7114.
* Version of the gateware and runtime loaded in the core device: Does not need the gateware.
* Hardware involved: No hardware.

dnadlinger, 2021-03-10:
Thanks for the great bug report, will have a look later this week.

dnadlinger, 2021-03-14:
I've created https://github.com/m-labs/artiq/issues/1632 for the sub-issue (unrelated to NumPy arrays) regarding having to use `length = self.n` instead of `self.n` directly.

closed 2021-03-14 by dnadlinger

dnadlinger, 2021-03-15:
The issue as filed should be taken care of, but please don't hesitate to create new bug reports for any other problems you might encounter – the NumPy array implementation was completely rewritten for ARTIQ 6 (arrays had previously been treated just the same as lists), so some issues are bound to have slipped through due to imperfect test coverage.

commit c707ccf7 (2021-03-14): compiler: Properly implement NumPy array slicing

Strided slicing of one-dimensional arrays (i.e. with non-trivial
steps) might have previously been working, but would have had
different semantics, as all slices were copies rather than a view
into the original data.

Fixing this in the future will require adding support for an index
stride field/tuple to our array representation (and all the
associated indexing logic).

GitHub: Fixes #1627.
commit 92501468 (2021-03-15): compiler: Properly implement NumPy array slicing

Strided slicing of one-dimensional arrays (i.e. with non-trivial
steps) might have previously been working, but would have had
different semantics, as all slices were copies rather than a view
into the original data.

Fixing this in the future will require adding support for an index
stride field/tuple to our array representation (and all the
associated indexing logic).

GitHub: Fixes #1627.