m-labs/artiq #1630: Phaser NCO does not generate correct frequency

ibes-26, 2021-03-10:
# Bug Report

## One-Line Summary
Numerically Controlled Oscillator (NCO) is not working.

## Issue Details
The NCO upconversion set_nco_frequency() does not change the generated frequency. Instead the frequency remains as it were with set_nco_frequency(0). No error is raised.
### Steps to Reproduce

````
from artiq.experiment import *
class PhaserHarmonics(EnvExperiment):
     def build(self):
        self.setattr_device("core")
        self.setattr_device("phaser_1")
        self.setattr_device("scheduler")
    @kernel
    def run(self):
        f = self.phaser_1
        self.core.reset()
        self.core.break_realtime()
        f.init()
        self.core.break_realtime()
        ch = f.channel[0]
        ch.set_att(0*dB)
        ch.set_duc_frequency(120*MHz)
        ch.set_nco_frequency(200*MHz)
        ch.set_nco_phase(0.)
        ch.set_duc_phase(0.)
        ch.set_duc_cfg(select=0, clr=0)
        delay(.1*ms)
        ch.oscillator[0].set_frequency(0)
        delay(0.1*ms)
        ch.oscillator[0].set_amplitude_phase(1.)
        delay(0.1*ms)
        f.duc_stb()
        pausecheck = 0
        while True:
            delay(50*ms)

            pausecheck += 1
            if pausecheck % 40 == 0:
                if self.scheduler.check_pause():
                    break
````

### Expected Behavior
tone at 320 MHz

### Actual (undesired) Behavior
tone at 120 MHz

### Your System (omit irrelevant parts)

Using Artiq master & Phaser master gateware / firmware.
ARTIQ version: ARTIQ v6.0.dev+66.ga8c5ad3b

jordens, 2021-03-11:
The nco and multipliers need to be enabled and the dac configured correctly. See the datasheet.

ibes-26, 2021-03-11:
> 
> 
> The nco and multipliers need to be enabled and the dac configured correctly. See the datasheet.

Thank you for the reply. Could you circulate the correct dictionary that needs to be passed in the device_db in order to enable the nco/ multipliers and to configure the dac?

cjbe, 2021-03-11:
@jordens can you post the code you used to test the NCO operation?

cjbe, 2021-03-11:
@ibes-26 it looks like we need, at a minimum:
```
nco_ena = 1
mixer_ena = 1
mixer_gain = 0
```
we may want to also add 3dB gain in the QMC

pathfinder49, 2021-03-17:
> The nco and multipliers need to be enabled and the dac configured correctly. See the datasheet.

After some time reading the source, am I understanding correctly that: 
1. The NCO needs to be configured through the `dac` dictionary argument in the device_db 
2. This code should work as expected when used with a suitably configured device_db

If so, we may avoid future support questions by mentioning this in the docs.

pathfinder49, 2021-03-31:
Comparing to the programming example in the data-sheet, there seems to be an error in the NCO register programming.

### In the datasheet
This NCO frequency...
[image]
... is programmed as ...
[image]
### coredevice implementation
In the coredevice driver, the most and least significant 16 bits seem to be swapped.
https://github.com/m-labs/artiq/blob/8002fcf8bbeb3425f19d66244130c6f1b1f9b7a3/artiq/coredevice/phaser.py#L760-L767

pathfinder49, 2021-03-31:
Beyond the swapped bits in `set_nco_frequency_mu()`, the NCO freqeuncy does not update when set via `set_nco_frequency()`. However, the NCO does work when set via the device_db. 
```Python3
        "dac": {
            "mixer_ena": 1,
            "mixer_gain": 1,
            "nco_ena": 1,
            # set NCO to 400 MHz
            "phase_addab_lsb": 26214,
            "phase_addab_msb": 26214,
            "phase_addcd_lsb": 26214,
            "phase_addcd_msb": 26214,}
```

pathfinder49, 2021-03-31:
Fixing #1650, I'm now able to test the `set_nco_frequency()` method. It seems like the bit-swap may be an error in the data-sheet. The current method works as intended (after clearing sif_sync manually). 

@jordens I assume the phaser gateware does not swap the NCO registers behind the scenes?

> Comparing to the programming example in the data-sheet, there seems to be an error in the NCO register programming.
> ### In the datasheet
>
> This NCO frequency...
> [image]
> ... is programmed as ... [...]

commit 75445fe5 (2021-05-12): Phaser: expose and automate clearing of DAC `sif_sync` (close #1630 and #1650)
`sif_sync` must be triggered to apply NCO frequency changes. To achieve per
channel frequency tunability exeeding the range of the DUC, the NCO frequeny must
adjusted. User code will need to trigger `sif_sync` to achieve this.

`sif_sync` can only be triggered if the bit was cleared. To avoid this pitfall,
the clearing of `sif_sync` is automated.
closed 2021-04-16 by jordens