m-labs/artiq #1631: artiq_run on Kasli not working on Windows 10 after upgrade to artiq 6.7582

drvolpe, 2021-03-11:
# Bug Report

## One-Line Summary

artiq_run not working with basic program on Kasli on Windows 10 after upgrade to artiq 6.7582. On artiq 5.7151 same program works.

## Issue Details

[ARTIQ.zip](https://github.com/m-labs/artiq/files/6123568/ARTIQ.zip)

### Steps to Reproduce

1. Run artiq_run not_working.py on a artiq 6.7 environment: the program should only display the print() statements in the build() (i.e., until "Flag 2"). Only the very minimal working.py executes correctly on artiq 6.7. When Ctrl-C is given to terminate the hung program, the following is output:

forrtl: error (200): program aborting due to control-C event
Image              PC                Routine            Line        Source
libifcoremd.dll    00007FFFCF043B58  Unknown               Unknown  Unknown
KERNELBASE.dll     00007FF84C22B443  Unknown               Unknown  Unknown
KERNEL32.DLL       00007FF84C7E7034  Unknown               Unknown  Unknown
ntdll.dll          00007FF84E7BD241  Unknown               Unknown  Unknown

2. Run artiq_run not_working.py on artiq 5.7151 and it works fine.
3. Actually, even a run() method with only a print() statement fails to execute if the at kernel decorator is present (as you can see in the screenshot below). Removing it makes the program just run fine.

### Expected Behavior

Should run.

### Actual (undesired) Behavior

Seems that the kernel cannot properly connect to the pc: see the following screenshot for what came out on the (USB) serial port:
[image]

### Your System (omit irrelevant parts)

* Operating System: Windows 10 Version 10.0.19042.804
* ARTIQ version: (with recent versions of ARTIQ, run ``artiq_client --version``) 6.7582
* Version of the gateware and runtime loaded in the core device: (in the output of ``artiq_coremgmt -D .... log``) gateware 6.7267.4340a5cf.beta
* If using Conda, output of `conda list` (please submit as a file attachment, as this tends to be long)
[conda_list_artiq_6.7582.txt](https://github.com/m-labs/artiq/files/6123694/conda_list_artiq_6.7582.txt)
[conda_list_artiq_5.7151.txt](https://github.com/m-labs/artiq/files/6123696/conda_list_artiq_5.7151.txt)
* Hardware involved: Kasli

sbourdeauducq, 2021-03-11:
The network protocols used between ARTIQ 5 and 6 are not compatible, and 
there are also some incompatibilities between ARTIQ-6 versions before 
the release (see the Git commit history for details). Are you keeping 
the core device firmware in sync with the software?

drvolpe, 2021-03-11:
The core device firmware is slightly out of sync wrt the software version: 6.7267 vs 6.7582. Do you think this is the problem?
Would you suggest flashing the same version on the Kasli to see if the problem disappears?

sbourdeauducq, 2021-03-11:
Yes

dnadlinger, 2021-03-11:
In particular, if the firmware version you are using still works with the ARTIQ 5 tools, it won't work with the ARTIQ 6 ones.

drvolpe, 2021-03-11:
OK, thank all of you very much! Tomorrow or Monday I'll flash it and, in case it works, I'll close the issue.

sbourdeauducq, 2021-03-21:
@drvolpe Is this resolved?

closed 2021-03-24 by sbourdeauducq