m-labs/artiq #1635: bool list converts to int list during RPC data transfer

lriesebos, 2021-03-17:
# Bug Report

## One-Line Summary

When sending a list of bool from the kernel to the host using an RPC function, the list arrives as a list of ints.
This happens since ARTIQ 6.
@sbourdeauducq is this expected behavior?

## Issue Details

### Steps to Reproduce

```python
from artiq.experiment import *

class ArtiqTestExperiment(EnvExperiment):

    def build(self):
        self.setattr_device('core')

    @kernel
    def run(self):
        self.foo([True])
        self.foo([False])

    def foo(self, val):
        print(f'val={val[0]}, type={type(val[0])}')
```

### Expected Behavior

Output of the program:
```
val=True, type=<class 'bool'>
val=False, type=<class 'bool'>
```

### Actual (undesired) Behavior

The values arrive as a list of integers. Output of the program:
Output of the program:
```
val=1, type=<class 'int'>
val=0, type=<class 'int'>
```

This behavior is only observed for bool lists. Sending a single bool value over RPC works as expected.

### Your System (omit irrelevant parts)

* Operating System: Ubuntu 20.04
* ARTIQ version: ARTIQ v6.7582.c2248278

sbourdeauducq, 2021-03-17:
>  @sbourdeauducq is this expected behavior?

No.

closed 2021-03-20 by dnadlinger

lriesebos, 2021-03-20:
Really appreciate the fix @dnadlinger !

commit 888696f5 (2021-03-20): coredevice: Fix RPC typing for bool lists/arrays

GitHub: Fixes #1635.
commit ca6db878 (2021-03-20): coredevice: Fix RPC typing for bool lists/arrays

GitHub: Fixes #1635.