m-labs/artiq #1636: change attribute of a user-defined object in the kernel

fanmingyu212, 2021-03-18:
# Question

Is there a way to modify an object from a user-defined class and to create a new object in the kernel code?

## Category: development

## Description

I cannot change the attribute `self.obj1.a` in the following code.
```python
from artiq.experiment import *

class CustomClass:
    def __init__(self):
        self.a = 1

    @portable
    def modify_object(self):
        self.a = 2

    @rpc
    def print_object(self):
        print(self.a)

class KernelReturnsObject(EnvExperiment):
    """Tests if a kernel function can return an object."""
    def build(self):
        self.setattr_device("core")
        self.obj1 = CustomClass()
        #self.obj1.modify_object()

    @kernel
    def run(self):
        self.core.reset()
        self.obj1.print_object()
        self.obj1.modify_object()
        self.obj1.print_object()

        #obj2 = CustomClass()
        #obj2.print_object()
```

When called in the kernel, `modify_object` function does not do anything (outputs `"1\n1"`). When called in the host (uncomment `self.obj1.modify_object()` line in the `build` function), the function works (outputs `"2\n2"`).

Also I cannot create an object in the kernel and use it. If the last two lines in the `run` function are uncommented, it raises the below exception
```
kernel_return_object.py:31:14-31:26: error: type <instance artiq_run_kernel_return_object.CustomClass {
                __objectid__: numpy.int32
        }> does not have an attribute 'print_object'
        obj2.print_object()
        ~~~~ ^^^^^^^^^^^^
```

I am not sure these are bugs or the expected behavior. Is there a way to modifying custom objects in the kernel?

sbourdeauducq, 2021-03-18:
> When called in the kernel, `modify_object` function does not do anything (outputs `"1\n1"`). When called in the host (uncomment `self.obj1.modify_object()` line in the `build` function), the function works (outputs `"2\n2"`).

That's because device values are not synchronized into the host at every RPC for performance reasons. The NAC3 proposal imposes restrictions on what values could be accessed from the host and device to avoid this confusion.

> Also I cannot create an object in the kernel and use it.

The current compiler cannot do this. NAC3 will likely support it.

closed 2021-03-18 by sbourdeauducq

fanmingyu212, 2021-03-18:
@sbourdeauducq thanks for the help.

> That's because device values are not synchronized into the host at every RPC for performance reasons.

Is there a way to force synchronization of the device / host values with the current version of artiq? In some cases, we want to save experimental results in an object and send the results (attributes of the object) to host to save data.

sbourdeauducq, 2021-03-18:
You can pass device values as parameters to an RPC.

sbourdeauducq, 2021-03-18:
Attributes are also synchronized when the kernel exits.