m-labs/artiq #1638: Phaser Upconverter LO frequency configuration

pathfinder49, 2021-03-18:
# Bug Report

## One-Line Summary

Phaser-Upconverter LO configuration seems to be missing options.

## Issue Details
I'm attempting to create a `device_db` for a phaser-upconverter. With the current coredevice driver the registers inside the TRF372017 must be configured manually. The documentations references the TRF372017 data-sheet without further explanation. 
https://github.com/m-labs/artiq/blob/d04bcd8754f11780d02d179285712980701d6518/artiq/coredevice/trf372017.py#L1-L6
To avoid misunderstandings, I'll summarise my understanding of the required configuration options. Currently, all bar one of these options seem to be exposed.

From the data-sheet it seems like the following values must be configured to achieve a desired LO frequency:
* LO_DIV_SEL
* PLL_DIV_SEL
* NINT (**this name does not appear in the driver**, @jordens am I missing something obvious?)
* PRSC_SEL
* RDIV 

For "fractional mode" the following additional options are used
* NFRAC and
[image]

If an external VCO/LO is provided the following options are needed:
* EN_EXTVCO
* NEG_VCO

### Expected Behavior
All the options above should be exposed and the names should match with the data-sheet. (Naming already matches for the exposed options)

jordens, 2021-03-18:
`ndiv`?

pathfinder49, 2021-03-18:
It looks like it does indeed correspond to NINT in your implementation.
https://github.com/m-labs/artiq/blob/d04bcd8754f11780d02d179285712980701d6518/artiq/coredevice/trf372017.py#L93-L97

Givent the documentation is the data-sheet, it seems best to match the names from the datasheet. Especially, as NDIV is used for something else in the data-sheet.
[image]

jordens, 2021-03-31:
No. That's exactly the same thing and not something else.
But do rename it.

commit 7404152e (2021-05-12): Phaser upconverter: rename `ndiv` -> `nint` to match datasheet (close #1638)
closed 2021-05-12 by jordens