m-labs/artiq #1639: Sayma: occasional failure on SYSREF DDMTD phase / RTIO TSC alignment

HarryMakes, 2021-03-19:
## One-Line Summary

Sayma AMC firmware would report failure for SYSREF trying to reach DDMTD phase target, and/or trying to align to RTIO TSC phase.

## Issue Details

Recently, since I've been testing Sayma for consistency across multiple power-cycles, I've observed that Sayma AMC can occasionally emit the following errors on the log upon power-cycling or reloading the FPGA (`artiq_flash start`):

```
ERROR(satman::jdcg::jesd204sync): failed to align SYSREF at FPGA: failed to reach SYSREF DDMTD phase target
```
```
ERROR(satman): failed to align SYSREF with TSC (SYSREF failed S/H timing)
```

These errors would happen randomly. Besides, the RTIO TSC alignment error might not even happen even after SYSREF fails to reach the DDMTD phase target. Due to my lack of readings into the code and DDMTD mechanism at this moment, I cannot tell whether or not the RTIO TSC is properly aligned when SYSREF reports DDMTD alignment errors.

I went further to have the firmware print the delta value when measuring the DDMTD phase, and usually it would fail when the minimum delta only reaches around 7, while it would pass at 0-2 normally. Again I currently don't know the implications yet.

<details>
<summary>Attachment: Example full log within a single power-cycle and with multiple FPGA reloads</summary>

*Note: this is a slightly-modified gateware based on c22482787ef806eb8e48d59b6ded7d93c7677c81 to fix some DRTIO transceiver errors and add a hardcoded SERDES TTL output driver.*
```
 __  __ _ ____         ____
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |
| |  | | |___) | (_) | |___
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2021 M-Labs Limited

Bootloader CRC passed
Gateware ident 6.ttlsqwave;satellite.syncdds
Initializing SDRAM...
DQS initial delay: 112 taps
Write leveling scan:
Module 3:
0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000010111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111000000000000000000000000000000000000000000000000000000000000
Module 2:
0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110000000000000000000000000000000000000000000
Module 1:
0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000010111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
0000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110111111100000000000000000000000000000000000000000000000000000000000000000000000000000000000
DQS initial delay: 112 taps
Write leveling: 80 102 118 114 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 249+-94 241+-98 213+-93 204+-91 done
SDRAM initialized
Memory test passed

Loading slave FPGA gateware...
  magic: 0x5352544d, length: 0x0019de0c
  DONE before loading
  ...done
Booting from flash...
Starting firmware.
[     0.000007s]  INFO(satman): ARTIQ satellite manager starting...
[     0.004753s]  INFO(satman): software ident 6.ttlsqwave;satellite.syncdds
[     0.011451s]  INFO(satman): gateware ident 6.ttlsqwave;satellite.syncdds
[... 127 lines cut ...]
[    92.560810s]  INFO(satman::jdcg::jdac): Running STPL test on DAC-1...
[    92.608298s]  INFO(satman::jdcg::jdac):   ...done STPL test
[    93.183455s]  INFO(satman): TSC loaded from uplink
[    93.186861s]  INFO(satman::jdcg::jesd204sync): aligning SYSREF with RTIO TSC...
[    93.197428s] ERROR(satman): failed to align SYSREF with TSC (SYSREF failed S/H timing)
[    93.203963s]  INFO(satman::jdcg::jesd204sync): resynchronizing DAC-0
[    93.211416s]  INFO(satman::jdcg::jesd204sync): resynchronizing DAC-1
[    93.308948s]  INFO(satman): rank: 1
[    93.311056s]  INFO(satman): routing table: RoutingTable { 0: 0; 1: 7 0; 2: 7 1 0; }
```

</details>

### Suggested Approach

Since we've seen issue surrounding UltraScale Xilinx instances before, I've tried to review the requirements for instances used for DDMTD on Sayma RTM (e.g. MMCM) and SYSREF sampler on Sayma AMC (e.g. ISERDESE3), but they seem to have been met by our code already. It is unlikely that the UltraScale instances are causing the occasional errors.

It might be possible that certain design flaws are present in the DDMTD core itself. From my early understanding of DDMTD, the down-converted clock signals <img src="https://latex.codecogs.com/gif.latex?\{u_{beat1},u_{beat2}\}" />  need to be deglitched by one of the following methods:
1. First Edge: select the first positive edge
2. Mean Edge: select the edge at the mean position at a bucket of timestamps
3. Zero Count: select the current edge when the same number of zeros and ones have been collected

The method chosen currently seems to be "first edge", as in the following code excerpt:
https://github.com/m-labs/artiq/blob/d04bcd8754f11780d02d179285712980701d6518/artiq/gateware/jesd204_tools.py#L157
While this method requires the simplest hardware implementation, the [original paper](https://www.ee.ucl.ac.uk/lcs/previous/LCS2011/LCS1136.pdf) seems to suggest that at lower frequencies other methods would yield a better time difference error. Perhaps the other two methods could yield better deglitched signals, but then the DDMTD core would need to be re-written carefully and tested.

sbourdeauducq, 2021-03-19:
There's already a self-test of the DDMTD core built into the firmware and run automatically, which should be able to detect problems due to glitches.

jbqubit, 2021-03-19:
@HarryMakes This paper has a bit more detail than the one you mentioned. 

https://white-rabbit.web.cern.ch/documents/DDMTD_for_Sub-ns_Synchronization.pdf

HarryMakes, 2021-11-25:
Hi all, I have a question regarding the difference between `AVG_PRECISION_SHIFT` and `SYSREF_SH_PRECISION_SHIFT` as shown below:

https://github.com/m-labs/artiq/blob/946254d22e2791099291c79c6907e041e6551666/artiq/firmware/satman/jdcg.rs#L259-L261

https://github.com/m-labs/artiq/blob/946254d22e2791099291c79c6907e041e6551666/artiq/firmware/satman/jdac_common.rs#L64-L66

*Values of `RAW_DDMTD_N_SHIFT`, `DDMTD_DITHER_BITS` and `DDMTD_N_SHIFT` for reference:*
https://github.com/m-labs/artiq/blob/946254d22e2791099291c79c6907e041e6551666/artiq/firmware/satman/jdac_common.rs#L33-L37

If my understanding is correct, `jdcg::jesd204sync::measure_sysref_sh_limits()` is relevant in SYSREF<>RTIO alignment for finding the SYSREF S/H average limits (in DDMTD phases). It measures DDMTD phase values via `jdac_common::measure_ddmdt_phase()`. Subsequently, the mean values are used for calibrating the target DDMTD phase for SYSREF for HMC7043 to slip ("coarse delay control").

While there doesn't seem to be connection between the precision required for measuring the DDMTD phases (`AVG_PRECISION_SHIFT`) and the precision for averaging the measured S/H limits (`SYSREF_SH_PRECISION_SHIFT`), is there a specific reason these two should be set up this way? To me, the array of measured values should be bitshifted to the left enough before averaging, if that makes sense. Thank you.