m-labs/artiq #1641: scheduler: check for mistyped / non-present arguments

charlesbaynham, 2021-03-23:
# ARTIQ Feature Request

## Problem this request addresses

Submitting an experiment via the scheduler device, it's possible to add incorrect arguments to your call. ARTIQ throws no warning / error, and will happily use the default values for the arguments that you thought you'd passed. 

For example, running `ScheduleSomething` in the following:

``` python
class DoSomething(EnvExperiment):
    def build(self):
        self.setattr_argument(
            "important_param", StringValue(default="Fire the missiles")
        )

    def run(self):
        print(self.important_param)

class ScheduleSomething(EnvExperiment):
    def build(self):
        self.setattr_device("scheduler")

    def run(self):
        new_expid = {
            "file": __file__,
            "class_name": DoSomething.__name__,
            "arguments": {"important_poram": "Abort the mission"},  # Oops, a typo
            "log_level": self.scheduler.expid["log_level"],
            "repo_rev": self.scheduler.expid["repo_rev"],
        }

        self.scheduler.submit("main", new_expid)
```

results in the missiles being fired without any error message / warning being displayed to the user. 

## Describe the solution you'd like

I'd suggest adding a check that all arguments passed are valid arguments and throwing an error if this is not the case. I can't think of a reason that passing extra unused parameters would be anything other than an error, but I may be wrong about that. 

To do this, the scheduler will probably need to call `worker.examine()` to get the default set of arguments and compare it with the passed ones. I can make a PR if people are happy with the approach.

hartytp, 2021-03-23:
Tangentially related to the above: I've always found the scheduler interface a bit opaque. As far as I'm aware (please do correct me if I'm wrong) the `expid` parameter isn't documented anywhere. Assuming the scheduler is intended to be public API rather than an internal, it would be great if it were documented. Or, perhaps replace `expid` in `scheduler.submit` with arguments and then have that method construct the dictionary itself (i.e. so we have an `arguments` kwarg which takes a dictionary).

charlesbaynham, 2021-03-23:
I agree, it definitely takes some digging to find the right structure for `expid`.

hartytp, 2021-03-23:
yes, IIRC last time I used it I ended up scheduling an experiment and then querying the schedule queue to find out the structure of the `expid` dictionary. Moving to a documented set of kwargs (and ideally adding some validation at the same time) would be relatively quick if we agree that it's the right thing to do (and that this is the right / only place that it needs to be done)...particularly if someone could give a bit more insight into the valid fields for `expid`

jbqubit, 2021-04-05:
See also discussion in the [Forum](https://forum.m-labs.hk/d/168-experiment-submitting-another-experiment).

lriesebos, 2021-04-20:
This might be off-topic, but in case you were investigating this because you want to run your experiments periodically, we have created some infrastructure for that. An [example is shown here](https://gitlab.com/duke-artiq/dax-example/-/blob/master/repository/scheduler/scheduler.py). That could save you the hassle of constructing the `expid` object, though it obviously does not fix the mistyped argument issue.

hanros1994, 2021-06-11:
> example is shown here

Thanks a lot for this awesome functionality, I was looking for something just like this but have been stuck on expid formatting. I have read through your dax.scheduler guide but I have a question do I add the controller key to the device_db.py file? 

[image]

lriesebos, 2021-06-14:
@hanros1994 Glad to hear we can help!

And yes, the controller key needs to be added to the device db, but only if you need its features. "Simple scheduling" works fine without. If added, other processes are able to influence the behavior of the scheduler by sending commands through this controller. See also our example project https://gitlab.com/duke-artiq/dax-example/-/blob/master/device_db.py#L365 .

I know our documentation is a bit outdated and incomplete. Feel free to open an issue in the DAX repo in case you have any questions, we're happy to help out!

hanros1994, 2021-06-14:
Thanks alot, I did have a few questions. I will post them in the DAX repo.

hartytp, 2021-07-15:
If the structure of `expid` is fixed, isn't this a situation where a `namedtuple` might make sense? Or, is the issue that pyon doesn't handle them?

Also, nitpicking, but I've always found the name `expid` quite confusing. It sounds like an `id` which should be a number, rather than a complete description of the experiment to be run.

PR #1927 (2022-07-26): language.environment: Add check_unused_arguments

# ARTIQ Pull Request

## Description of Changes
**Problem**: scheduler device might have added incorrect arguments as stated in issue #1641

**Solution**: Check the unused arguments in expid in unprocessed_arguments after build

**Details**: remove argument in unprocessed_arguments after get() during build. Call check_unused_arguments() after build to see what argument is left. Raise exception and state those arguments.

**Test result**:
[image]
Based on the example in #1641 
After submitting RID 40, exception occur and stating what arguments w [... cut]
commit 2a7a72b2 (2022-07-26): language.environment: error out if unknown arguments are passed (#1927)

Closes #1641
closed 2022-07-26 by sbourdeauducq