m-labs/artiq #1647: Phaser spontaneously enters lockup state requiring power-cycle

pathfinder49, 2021-03-29:
# Bug Report

## One-Line Summary

Phaser can spontaneously enter a locked-up state that requires a power cycle. This may occur on initial power-up or spontaneously lock-up at a later point. 

## Issue Details
When Phaser has entered a locked-up state, runnign `Phaser.init()` results in `ValueError: DUC+oscillator phase/amplitude test failed`. This state is irrecoverable short of a power cycle. Restarting via flashing is not sufficient. This seems very similar to #1597. However, no interrupted/failed experiments are involved.

### Steps to Reproduce

1. Setup a Kasli in a crate with Phaser
2. Power the system
3. run `Phaser.init()`
4. If initialisation succeed, wait about 15 minutes and go back to 3. (Alternatively power cycle repeatedly until initialisation fails)

### Your System (omit irrelevant parts)

* ARTIQ version: Most recent artiq 7.0
* Version of the gateware and runtime loaded in the core device: Most recent artiq 7.0
* Hardware involved: Kasli 1.1, Phaser 1.0 (upconverter variant)

jordens, 2021-03-30:
I can't reproduce that with repeated `init()`.
You don't say which version of the phaser gateware this is.

jordens, 2021-03-30:
Please provide an MWE that reproduces this for you.

pathfinder49, 2021-04-01:
Apologies for not including the phaser gateware in my initial report. This is the most recent release at the time of opening the issue (v0.4).

Here the MWE:
```Python3
from artiq.experiment import *

class Test(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("phaser")

    @kernel
    def run(self):
        self.core.wait_until_mu(now_mu())
        self.core.break_realtime()
        self.phaser.init(debug=False)
```

I see that another release has been published today. I'll see if its still an issue there.

jordens, 2021-04-01:
Well yes. We didn't get feedback on whether the changes fix https://github.com/quartiq/phaser/issues/5 for you guys or not and were waiting with a release.

pathfinder49, 2021-04-07:
The release v0.5 seems to have fixed this :+1:

closed 2021-04-07 by pathfinder49