m-labs/artiq #1650: Phaser:  DAC `sif_sync` not cleared in `init()`

pathfinder49, 2021-03-31:
# Bug Report

## One-Line Summary

During `init()` the sif_sync bit of the Phaser DAC is set and not cleared. This blocks further sync until manually cleared.

## Issue Details
The `Phaser.init()` method sets the `sif_sync` bit. This is required to apply the written NCO values. Unless the bit is cleared, any changes to the NCO frequency can not be applied.
[image]

Currently, this bit is set when applying the DAC memory map (as defined in `artiq.coredevice.dac34h84`). This is not a suitable interface for syncing the NCO from user code.

### Expected Behavior
The bit should be cleared during `Phaser.init()`. Once the bit is cleared, the NCO frequency can be written using the `set_nco_frequency` method. 

A reasonable approach would be to provide a `sync_nco` method that applies written NCO changes. This would also avoid the need for manually accessing the DAC registers to update the NCO frequency. Currently this has to be done as shown below:
```Python
        data = self.phaser.dac_mmap[-9]
        data0 = data & ~int32(0b10)
        data1 = data
        self.phaser.dac_write(data0 >> 16, data0)
        self.phaser.dac_write(data1 >> 16, data1)
```

### Actual (undesired) Behavior
The bit is not cleared during `Phaser.init()`. This blocks further syncing of NCO frequency changes, until the bit has been manually cleared.

jordens, 2021-03-31:
From the datasheet I understood `sif_sync` as one-shot rising edge sensitive. If that's correct, there is no point in clearing it since you have to do two writes for each sync anyway and this issue is invalid.

jordens, 2021-04-16:
It doesn't need to be cleared.

closed 2021-04-16 by jordens