m-labs/artiq #1651: Phaser initialisation leaves Kasli oscilator enabled

pathfinder49, 2021-03-31:
# Bug Report

## One-Line Summary

`Phaser.init()` sets the `phaser.channel[i].oscillator[0]` to full scale, and does not disable it again.

## Issue Details
During initialisation, phaser performs tests of each channel using the (Kasli) 25 MS/s NCO. For this `phaser.channel[i].oscillator[0]` is written to full scale. Thhi oscillator is not disabled/set to 0 amplitude again during initialisation. At the end of initialisation dac_tx is enabled. As the oscilator has not been disabled, this applies continuous full scale RF the the Phaser outputs. (Confirmed on hardware)
https://github.com/m-labs/artiq/blob/fc59791583905f9e35a4ef0bc997c21ebc649a64/artiq/coredevice/phaser.py#L288-L295

I see two points of concern:
1. The user does not expect phaser to be continuously producing RF after initialisation. 
2. Phaser may plausibly drive loads that are damaged by continuous full power (or inappropriate output frequency). RF output should only be enabled when needed/requested by the user.

jordens, 2021-03-31:
No. It doesn't produce rf but dc. Nothing is in danger. Nothing gets beyond the first decoupling cap or the xfmr. There are in fact two things that the user needs to do to get anything. Set a frequency and remove the clear bit. Alternatively configure duc or cmix or fmix. You actively have to deviate from the default configuration.

jordens, 2021-03-31:
Clearing the default amplitude to zero does make getting anything out of phaser one bit more complicated and is another hurdle for the user to clear. Urukul also defaults to full scale dc output, the same way as phaser.

pathfinder49, 2021-03-31:
> No. It doesn't produce rf but dc. Nothing is in danger. Nothing gets beyond the first decoupling cap or the xfmr. There are in fact two things that the user needs to do to get anything. Set a [...]

It does produce RF in either of the following scenarios:
1. `Phaser.init` is run to re-initialiise Phaser after previous use. In this case the DUC and Kasli NCOs will remain set at their last freqeuncy. Both will result in RF DAC output.
2. The DAC coarse mixer is configured via the device-db. The documentation gives the impression this is an intended usecase 
https://github.com/m-labs/artiq/blob/fc59791583905f9e35a4ef0bc997c21ebc649a64/artiq/coredevice/phaser.py#L83-L87
https://github.com/m-labs/artiq/blob/fc59791583905f9e35a4ef0bc997c21ebc649a64/artiq/coredevice/phaser.py#L111-L116

> 
> 
> Clearing the default amplitude to zero does make getting anything out of phaser one bit more complicated and is another hurdle for the user to clear. Urukul also defaults to full scale dc output, the same way as phaser.

Unlike Phaser, Urukul has an RF switch that must be opened by the user. As phaser does not have an output switch, the amplitude should default to zero.

pathfinder49, 2021-04-01:
> No. It doesn't produce rf but dc. Nothing is in danger. Nothing gets beyond the first decoupling cap or the xfmr. There are in fact two things that the user needs to do to get anything. Set a [...]

Ack, I had missed the significance of `clr=1` being set in the DUC and kasli-NCO. This means output is only enabled when either `oscillator.set_amplitude_phase` (expected) or `set_duc_cfg` (surprising) are called. I had called `set_duc_cfg` in my testing without the DAC-mixer. 

From the documentation and reading the source I had understood `set_duc_cfg` to be primarily intended to set phase coherence, not enable RF output, especially if `duc_stb` has not been called. For experienced artiq users this is an accident waiting to happen.
https://github.com/m-labs/artiq/blob/fc59791583905f9e35a4ef0bc997c21ebc649a64/artiq/coredevice/phaser.py#L710-L718

https://github.com/m-labs/artiq/blob/fc59791583905f9e35a4ef0bc997c21ebc649a64/artiq/coredevice/phaser.py#L458-L465

The `set_duc_cfg` behaviour is not clear from the method description. Further, using the DAC mixer results in full power RF output.

pathfinder49, 2021-04-01:
In addition to unexpected RF output, leaving oscillator 0 enabled at full scale (DC) makes the remaining Kasli oscillators unusable. Unless the oscillator 0 amplitude is written to a different amplitude, enabling any of the other oscillators results in severe clipping.

jordens, 2021-04-01:
Deviations from the default configuration can clearly cause RF to be emitted after init. Yes.
The DUC is cleared and CMIX and FMIX are disabled by default. By default there is no RF output.
What the general behavior independent of the configuration should and can be is certainly debatable.

I can only read most of your posts as tangential rants indicating some level of general frustration with things that aren't the way you think they should be. E.g. the trivial complaint that the addition of multiple signal sources can "severly" clip. That's basic digital data 101, abundantly documented for inexperienced users without any mental model of digital addition. It's universally true but very much irrelevant here.

In your approach to filing these issues rapid-fire I would like to request that you try to be a bit more open minded. It sets you up for disadvantaged confrontation if you start by postulating the outputs to your inputs and that other outputs are wrong. If the expected behavior isn't spelled out before, then the actual behavior isn't wrong. Constructing an expected behavior ex post is very subjective. One can argue about the "best" behavior, but it's much harder to argue about correctness. Nor should you assume that your interpretation of why things are as they are, what's relevant and what not, how things work, or how things should be fixed, is necessarily correct.
Rather doing the legwork of describing your setup, experiment, and observations *in full detail* and discussing possible expectations before demanding implementation changes according to your prescription would make it possible and enjoyable to give constructive and helpful responses.

pathfinder49, 2021-04-01:
I also prefer not to work and document the operation of phaser via issues and would prefer a more mutually constructive approach. I'd like to apologise if I come across as being set on a particular solution. When I suggest solutions, they represent what I believe to be an appropriate solution. 

Phaser is clearly a more complicated design than most EEMs, both on the hardware and software side. Within the Oxfrod group, I am (at least) the fifth person to try using Phaser. Unfortunately, the previous users were presented with repeated unexpected behaviour. This resulted in them abandoning the use of Phaser. Many of the issues I posed have previously been encountered (an not understood) by reasonably experienced ARTIQ users. ARTIQ is a control system for quantum mechanics experiments. Quantum Physicists are not experts in digital signal processing. There clearly is a large gap between many ARTIQ users available time and knowledge to that required to currently use Phaser. To me, opening an issue fulfils multiple functions that are useful to improving the usability of Phaser. They represent a resource where other users can see and better understand the quirks of the current implementation. They indicate discrepancies between the implementation and the behaviour an engaged and reasonably experienced ARTIQ user may expect. Finally, they act as a starting point for further documentation/implementation work on the Phaser core-device driver.

jordens, 2021-04-01:
If you want to work with qubits, you better be good at bits. That should be obvious. If you're surprised about the documented overflow/clipping behavior of digital signal source addition, or if you reduce yourself to being just a Quantum Physicist who knows his Bloch equations, then it's questionable whether "experienced ARTIQ user" is the right title. Wanting a high performance and complex digital signal processor but then being surprised that using it involves a lot of digital signal processing is naive at best. Yes, you don't need to understand how a car works to do some basic driving, and you don't need to understand how a signal generator works to use its basic functions. But this isn't the beetle of signal generators doing a single CW tone.
Phaser is complex, that was the design goal, proudly stressed. Not thinking about specifications ahead of time and instead expecting others to guess them for you is the perennial mistake that exacerbates the mismatch problem.
It's a common fallacy that what people think they want must therefore correct, good, or possible. This is beautifully demonstrated with the clipping/overflow behavior.
Opening issues alone does very little towards improving usability if it isn't followed through and acted on. Phaser, like any part of ARTIQ and Sinara, needs sustained maintenance and long term development.

commit 4f812cc4 (2021-05-12): Phaser: zero oscillator amplitude after `init()` (close #1651)
Currently, `init()` leaves a single oscillator at full scale. The phase
accumulator of this oscillator is held continuously cleared. Provided no
upconverting mechanism is active (DUC, CMIX, NCO), this produces a full-scale
DC voltage. The DC voltage is blocked by hardware capacitors. This behaviour
is not mentioned by the `init` documentation.

If one attempts to use any other oscillator without reducing the amplitude
of the oscillator enabled by `init`, there is by significant clipping.

In the case that the NCO or CMIX are config [... cut]
closed 2021-05-12 by jordens