m-labs/artiq #1653: Tool to Check the Firmware Version of Urukul

rgresia, 2021-04-07:
# ARTIQ Feature Request

## Problem this request addresses

#1652 

## Describe the solution you'd like

A tool or an addition to an existing tool to be able to determine the current firmware version of Urukul and possibly other devices. Another solution could be an automated check that alerts the user `current firmware: X  on Y device is  incompatible`.

## Additional context

Earlier versions of Urukul, v1.2-3, requires a specific [patch](https://github.com/sinara-hw/Urukul/wiki/patch2019) and there are a variety of [firmware versions](https://github.com/quartiq/urukul/releases) for Urukul CPLD.

sbourdeauducq, 2021-04-07:
> Earlier versions of Urukul, v1.2-3, requires a specific [patch](https://github.com/sinara-hw/Urukul/wiki/patch2019)

What does this have to do with the Urukul CPLD gateware?

sbourdeauducq, 2021-07-05:
> Another solution could be an automated check that alerts the user `current firmware: X on Y device is incompatible`.

AFAIK that's what the ``proto_rev`` check does on Urukul.

closed 2021-07-05 by sbourdeauducq

pathfinder49, 2021-07-05:
IIRC, `proto_rev` is shared between protocol-compatible CPLD versions. (Not quite the same as the firmware version)