m-labs/artiq #1655: RPCs corrupt integers in lists with little warning

pathfinder49, 2021-04-08:
# Bug Report

## One-Line Summary

RPCs returning a `TList(TIntXX)` return corrupted data for integers outside the IntXX range

## Issue Details

Retruning a list of integers from an RPC results in data corruption if any integers are outside the range of the integer returned by the RPC. This is the case for both `Int32` and `Int64`. This event is indicated by a message logged on debug level. 

The relevant message reads:
`artiq.coredevice.comm_kernel:rpc service: 3 [<artiq_worker_rpc_ints.RPCIntCorruption object at 0x7f1a7756d940>] {} ! error("'l' format requires -2147483648 <= number <= 2147483647")`

The analagous situation when returning Integers form RPCs is handled much more verbosely. Returning out-of-range integers from RPCs raises an exception:
`root:Terminating with exception (RPCReturnValueError: type mismatch: cannot serialize 2147483648 as 32-bit int`

Related comment from @sbourdeauducq 
https://github.com/m-labs/artiq/pull/1628#issuecomment-816298822

### Steps to Reproduce
Run the Following example:
```Python3
from artiq.experiment import *

class RPCIntCorruption(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    def int_from_rpc(self) -> TList(TInt32):
        number = 1<<31
        print(number)
        return [number]

    @kernel
    def run(self):
        print(self.int_from_rpc())
```

This prints:
```Python3
print:2147483648
print:[1515870810]
```
### Expected Behavior

The user should be alerted. Raising an exception, or minimally logging a warning seem appropreate.

### Your System (omit irrelevant parts)

* ARTIQ version: Latest Artiq

stevefan1999, 2021-12-16:
The reason for this is, there is a silent bug in comm_kernel.py:CommKernel::_send_rpc_value
https://github.com/m-labs/artiq/blob/9e5e234af31a82d053bc6c1add9eedc8bcf1d99b/artiq/coredevice/comm_kernel.py#L478-L486
Since the int value `1<<31` or 2147483648 exceeds 32-bit signed integer range, the `struct.pack` call effectively fails and so is `self._write`, and of course why the kernel read the stack canary since the values aren't sent to the kernel at all! I have proposed a quick fix for this but this require some more supervisions first.

sbourdeauducq, 2022-01-11:
Fixed in d7dd75e833fc99ad75e10220a5ec8f03d64ee5bd

PR #1801 (2022-01-12): comm_kernel: only send replies when no user-defined RPC function throws in service
# ARTIQ Pull Request

## Description of Changes

This PR fixes an issue when an exception happens after running the service function in RPC service routine. Previously I tried to catch `struct.error` as a quick fix, but it seems like the fundamental issue is not just because of that. 

The problem started with this part, after running the service routine:
https://github.com/m-labs/artiq/blob/bbac4770926048b3c0b434aede371a63d198ecd5/artiq/coredevice/comm_kernel.py#L561-L564

And when an exception happened around  [... cut]
closed 2022-01-11 by sbourdeauducq