m-labs/artiq #1656: Phaser output power depending on DUC frequency

Jacky2k, 2021-04-13:
# Bug Report

## One-Line Summary

The output power of phaser seems to depend on the DUC frequency in a non-expected way.

## Issue Details
There was a discussion on the output power of phaser in https://github.com/sinara-hw/Phaser/issues/130, which was basically resolved. During this process I discovered that the output power of Phaser depends on the DUC frequency in a non-expected way, as seen in the image below. The image shows the output power in dBm vs. the DUC frequency using an osciallator at 7 MHz and an LO for the upconverter at 2875 MHz.
[image]
The falloff at the end is expected due to the low-pass filter in the baseband, but I would expect the amplitude to be constant or rise towards lower DUC frequency values. @pathfinder49 suggested that this is a software issue and asked me to open an issue here - I don't understand why this might be a software issue.

### Steps to Reproduce
1. Run the script attached for different DUC frequencies (line 41) and measure the output power of channel 0.
[phaser_duc_minimal.zip](https://github.com/m-labs/artiq/files/6303335/phaser_duc_minimal.zip)

### Expected Behavior
I would expect a monotonically drop of the output power for higher DUC frequencies, plus maybe some attenuation at very low frequencies due to AC coupling of the baseband signal.

### Actual (undesired) Behavior
The output power rises between DC and ~180 MHz, which is uncommon for a low pass filtered signal.

### Your System (omit irrelevant parts)

* Operating System: Ubuntu 20.04
* ARTIQ version: ARTIQ v7.7602.fc597915.beta
* Version of the gateware and runtime loaded in the core device: software ident 7.7602.fc597915.beta;phaser, gateware ident 7.7602.fc597915.beta;phaser
* Hardware involved: Kasli + Phaser v1.0

jordens, 2021-04-13:
Highly unlikely this has anything to do with the duc as you claim. As requested months ago, characterization of the reconstruction filter is missing and should be your first priority.

pathfinder49, 2021-04-13:
@Jacky2k Thank you for the report :smiley: The DUC has a 200 MHz bandwidth. https://github.com/m-labs/artiq/blob/78490bef5d894e610bf4555f5b03071dc2115810/artiq/coredevice/phaser.py#L731-L737

I would therefore expect an output power drop around 200 MHz. I have not yet investigated the DUC gain curve. The DUC was implemented by @jordens on the Phaser FPGA. He should be the best person to asses the likelihood of this being related. I will report back here after looking at the reconstruction filter response using the DAC NCO.

pathfinder49, 2021-04-13:
Aside: My request to discuss this here was meant with regards to explanations of the digital signal path. Not the details of the entire gain curve.

pathfinder49, 2021-04-13:
@Jacky2k I've measured the power-frequency relation using the DUC and DAC-NCO. As expected, the DUC power drops off sharpy after 200 MHz.

The remaining power-frequency dependency is partially due to the Reconstruction filter https://github.com/sinara-hw/Phaser/issues/119#issuecomment-818800162. There fill be further contributions from the DAC and TRF gain.

The measurements below were performed with: 
* DAC mixer enabled
* DAC `mixer_gain=0`
* The DAC NCO enabled
* DAC-QMC enabled
* LO set to 2.875 GHz

[image]
[image]

closed 2021-04-13 by jordens