m-labs/artiq #1668: discussion: storing links to example code / user-submitted resources

charlesbaynham, 2021-04-29:
# ARTIQ Feature Request

Yesterday I mentioned a generic NDSP for writing to InfluxDB - I've tidied up the code which is available here: https://gitlab.com/charlesbaynham/artiq_influx_generic. @jonathanpritchard I think you mentioned interest. 

Where is the best place to gather links to things like this? The "List of available NDSPs" in the docs? The resources page on the M-Labs website at https://m-labs.hk/experiment-control/resources/ ? A sticky post on the forum? A wiki attached to this repository?

One of the things that came out of the discussion yesterday was a desire for more working examples for people to look at, "warts and all". These probably do not belong e.g. on the resources section of the M-labs website, but they're still a useful thing to have. How can we best organise contributions like these to encourage people to easily make them, without it becoming a cluttered mess somewhere?

## Suggestion

A possible approach would be to have a "Contributions" channel in the forum. That way anyone could submit a link and a brief description to their code, and there's a natural place for people to discuss it. Additionally, if that code is something that others could possibly use (like a framework or driver), it could also be added to the M-Labs website resources section and/or a page in the Sphinx docs. A sticky post at the top of the channel could be maintained to have links to the various submissions, organised by type.

sbourdeauducq, 2021-04-29:
> The "List of available NDSPs" in the docs? The resources page on the M-Labs website at https://m-labs.hk/experiment-control/resources/ ?

Yes, these two locations and feel free to advertise on the forum as well.

> These probably do not belong e.g. on the resources section of the M-labs website,

Why not? It's already a section called "Community code". Maybe there could be subsections such as "Example experiments", "NDSPs",...?

> A possible approach would be to have a "Contributions" channel in the forum. 

I have added that too.

jonathanpritchard, 2021-04-30:
Awesome thanks Charles, that is really useful - will see if we can get that implemented next week.

Do you want to add this as the first post in the "contributions thread"? Thanks for setting up Sebastian

charlesbaynham, 2021-05-04:
Thanks @sbourdeauducq , I've gone ahead and submitted it as https://forum.m-labs.hk/d/184-data-ndsp-driver-for-fast-writing-to-an-influxdb-time-series-database. Would you mind adding it to the resources page? I'll make PR for the docs NDSP page.

sbourdeauducq, 2021-05-10:
Done.