m-labs/artiq #1669: Type inference error for list of zero length

fanmingyu212, 2021-05-03:
# Bug Report

## One-Line Summary

Type of list of length zero cannot be fully inferred in the kernel.

## Issue Details

See https://github.com/m-labs/artiq/issues/1626 for an related issue. The linked issue has this type inference error for a numpy.array of length zero, but this issue is for a list of length zero.

### Steps to Reproduce

Running the following code reproduces the error.

```python
from artiq.language import *

class Test(EnvExperiment):
    def prepare(self):
        pass

    def build(self):
        self.setattr_device("core")
        self.scan_build = []

    @kernel
    def run(self):
        for kk in self.scan_build:
            kk.run_something()
```

### Expected Behavior

Experiment runs with no error.

### Actual (undesired) Behavior

It raises the following error during compiling.
```
test.py:13:13-13:15: error: the type of this expression cannot be fully inferred
        for kk in self.scan_build:
            ^^
test.py:13:13-13:15: note: the expression has type 'a
        for kk in self.scan_build:
            ^^                   
```
### Your System (omit irrelevant parts)

* Operating System: Windows 10 x64
* ARTIQ version: Latest master (https://github.com/m-labs/artiq/commit/47bf5d36afca350f1c69cebc74c1cf887c9f762c)

dnadlinger, 2021-05-03:
For lists, this is expected behaviour. How would the compiler figure out what the element type should be?

fanmingyu212, 2021-05-03:
@dnadlinger thanks for the response. I understand this problem cannot be solved without type annotations for attributes now.

One of the workarounds might be having one placeholder element in the list, such as in the code below:

```python
from artiq.language import *

class TestObj:
    def run_something(self):
        pass

class Test(EnvExperiment):
    def prepare(self):
        pass

    def build(self):
        self.setattr_device("core")
        self.scan_build = [TestObj()]

    @kernel
    def run(self):
        for kk in range(len(self.scan_build)):
            if kk > 0:
                self.scan_build[kk].run_something()
```

Not a elegant solution but it should work.

closed 2021-05-12 by sbourdeauducq

pca006132, 2021-10-30:
Just wondering if it is possible to have a non-empty list that we cannot infer the type. If not, maybe we can just skip generating the for loop if the element type is unknown (without having to add an empty list type to the system...)

fanmingyu212, 2021-10-30:
@pca006132 I guess a list consisting of elements of different types is a nonempty list that we cannot infer the type? In this case an exception likely should be raised (which is the current behavior) rather than quietly skipping the for loop.

pca006132, 2021-10-30:
> @pca006132 I guess a list consisting of elements of different types is a nonempty list that we cannot infer the type? In this case an exception likely should be raised (which is the current behavior) rather than quietly skipping the for loop.

Sorry, maybe I should make it precise: By cannot infer the type, I mean we cannot infer a concrete type for the list element (i.e. the result is a type variable). The case you mentioned should be a unification error (trying to unify values of incompatible types) instead of cannot infer a concrete type.

fanmingyu212, 2021-10-30:
Ah I see. Thanks for the clarification. It seems to me that what you suggested is a promising way to solve this problem.

PR #1670 (2021-05-12): manual: cannot use empty lists in kernel

# ARTIQ Pull Request

## Description of Changes

Adds a note in the `Compiler` section about using empty lists in the kernel.

### Related Issue

Closes #1669.

## Type of Changes

|   | Type |
| ------------- | ------------- |
| ✓  | :scroll: Docs |

## Steps (Choose relevant, delete irrelevant before submitting)

### All Pull Requests

- [x] Use correct spelling and grammar.
- [x] Close/update issues.
- [x] Check the copyright situation of your changes and sign off your patches (`git commit --signoff`, see [copyright](../CONTRIBUTING.rst#copyright-an [... cut]