m-labs/artiq #1671: Tuples of mixed types do not compile

filipmazurek, 2021-05-04:
# Bug Report

## One-Line Summary

Returning a TTuple containing mixed types from the kernel results in a compilation error.

## Issue Details

### Steps to Reproduce

The following code will trigger the compilation unification error. 

```
from artiq.experiment import *
import numpy as np

class ArtiqTestExperiment(EnvExperiment):

    def build(self):
        self.setattr_device('core')

    def run(self):
        self.return_tuple()

    @kernel
    def return_tuple(self) -> TTuple([TInt32, TBool]):
        return np.int32(1), True
```

### Expected Behavior

Expect the program to compile and allow mixed tuple types. The program works when returning a tuple of multiple of the same type or `TTuple([TInt32, TInt64])`

### Actual (undesired) Behavior

The result is a compilation unification error. This happens when attempting to mix TInt32 and TBool, TInt32 and TFloat, and likely other untested combinations. However, combining TInt32 and TInt64 works as expected.

```
Traceback (most recent call last):
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/artiq/compiler/transforms/inferencer.py", line 51, in _unify
    typea.unify(typeb)
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/artiq/compiler/types.py", line 91, in unify
    self.find().unify(other)
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/artiq/compiler/types.py", line 138, in unify
    raise UnificationError(self, other)
artiq.compiler.types.UnificationError: (artiq.compiler.types.TInstance('artiq.compiler.builtins.TBool', OrderedDict([('__objectid__', artiq.compiler.types.TMono('int', OrderedDict([('width', artiq.compiler.types.TValue(32))])))])), artiq.compiler.types.TInstance('artiq.compiler.builtins.TFloat', OrderedDict([('__objectid__', artiq.compiler.types.TMono('int', OrderedDict([('width', artiq.compiler.types.TValue(32))])))])))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/nix/store/i1w7skm7mv82grc1pn8m4vn41b4ka6xk-python3.8-artiq-6.7602.ec4270fb/bin/.artiq_run-wrapped", line 9, in <module>
    sys.exit(main())
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/artiq/frontend/artiq_run.py", line 225, in main
    return run(with_file=True)
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/artiq/frontend/artiq_run.py", line 211, in run
    raise exn
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/artiq/frontend/artiq_run.py", line 204, in run
    exp_inst.run()
  File "repository/tuple_test.py", line 14, in run
    self.return_tuple_5()
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/artiq/language/core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/artiq/coredevice/core.py", line 129, in run
    self.compile(function, args, kwargs, set_result)
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/artiq/coredevice/core.py", line 105, in compile
    stitcher.finalize()
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/artiq/compiler/embedding.py", line 705, in finalize
    inferencer.visit(self.typedtree)
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/pythonparser/algorithm.py", line 39, in visit
    return [self.visit(elt) for elt in obj]
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/pythonparser/algorithm.py", line 39, in <listcomp>
    return [self.visit(elt) for elt in obj]
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/pythonparser/algorithm.py", line 41, in visit
    return self._visit_one(obj)
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/pythonparser/algorithm.py", line 32, in _visit_one
    return getattr(self, visit_attr)(node)
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/artiq/compiler/transforms/inferencer.py", line 1680, in visit_FunctionDefT
    self.generic_visit(node)
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/pythonparser/algorithm.py", line 27, in generic_visit
    self.visit(getattr(node, field_name))
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/pythonparser/algorithm.py", line 41, in visit
    return self._visit_one(obj)
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/pythonparser/algorithm.py", line 32, in _visit_one
    return getattr(self, visit_attr)(node)
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/artiq/compiler/transforms/inferencer.py", line 1275, in visit_CallT
    self.generic_visit(node)
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/pythonparser/algorithm.py", line 27, in generic_visit
    self.visit(getattr(node, field_name))
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/pythonparser/algorithm.py", line 39, in visit
    return [self.visit(elt) for elt in obj]
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/pythonparser/algorithm.py", line 39, in <listcomp>
    return [self.visit(elt) for elt in obj]
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/pythonparser/algorithm.py", line 41, in visit
    return self._visit_one(obj)
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/pythonparser/algorithm.py", line 32, in _visit_one
    return getattr(self, visit_attr)(node)
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/artiq/compiler/transforms/inferencer.py", line 108, in visit_ListT
    self._unify(node.type["elt"], elt.type,
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/artiq/compiler/transforms/inferencer.py", line 74, in _unify
    diag = diagnostic.Diagnostic("error",
  File "/nix/store/3f7hxzbmylccbvjjr18ba5lxrgm3182q-python3-3.8.8-env/lib/python3.8/site-packages/pythonparser/diagnostic.py", line 53, in __init__
    raise ValueError("location and highlights must refer to the same source buffer")
ValueError: location and highlights must refer to the same source buffer
```

### Your System (omit irrelevant parts)

* Operating System: Ubuntu 20.04
* ARTIQ version: ARTIQ v6.7602.ec4270fb

lriesebos, 2021-07-23:
Just to add, this error occurs **only** when `TTuple` is used in the annotation of a return value. When used to annotate function arguments, there seems to be no problem. Also, not typing the return value (i.e. fully rely on type inference of the compiler) seems to work fine and is probably the easiest workaround.

stevefan1999, 2021-12-20:
Should this AST be correct?
[image]

dnadlinger, 2023-09-26:
Should be fixed in https://github.com/m-labs/artiq/pull/2164.

closed 2023-09-26 by dnadlinger