m-labs/artiq #1674: master / worker ipc: possible 10x speed increase

charlesbaynham, 2021-05-10:
# ARTIQ Feature Request

## Problem this request addresses

Transferring datasets between master and worker can be slow. For example, transferring 1 million floats currently takes 5700ms on my PC. This is a local transfer, not involving sypico's sync_structs and network capabilities, only using asyncio streams. 

## Describe the solution you'd like

Much of this time is spent in the serialization and deserialization of data. By using Pickle instead (which is powered by compiled c) you can get about a factor of 10x improvement. 

In PR xxx I've added a unit test to check the speed the ipc comms. Calling it with e.g. `pytest artiq\test\test_worker.py -rP` I get the performance mentioned above. Calling it again after applying all the changes in the PR gives 450ms for a 1 million float transfer: a speedup of ~10x. 

[image]

## Additional context

Before I mark the PR as ready for review, are there any considerations here which need thinking about?

sbourdeauducq, 2021-05-10:
For IPC I don't see an issue since both ends always come from the same software package AFAIK.

sbourdeauducq, 2021-05-10:
Does msgpack support all the data types we want?

charlesbaynham, 2021-05-10:
Nope, that's another drawback. Packages exist to add (fast) numpy support, but it has the same issues with tuples as JSON does. It's basically a binary version of JSON. So that'd need a solution, but it sounds solvable. 

If we were considering a rethink of sypico, I think ZeroMQ might be another good thing to consider (https://zguide.zeromq.org/docs/preface/). Whilst it does include some binary serialisation, it's really aimed at being a lightning-fast IPC / networking socket provider. That's a PR for another day, however.

jordens, 2021-05-10:
+1 on using existing MQ approaches (having been a long term user) for certain use cases.

airwoodix, 2021-05-10:
> Does msgpack support all the data types we want?

As already mentioned, there's [msgpack_numpy](https://pypi.org/project/msgpack-numpy/). For the other types that are not supported out of the box (e.g. ``slice``, ``tuple``, ``set``) one could use the [Ext type](https://github.com/msgpack/msgpack-python#extended-types).

> If we were considering a rethink of sipyco, I think ZeroMQ might be another good thing to consider

+1 Like for msgpack, it would also easily provide better documentation / support thanks to their much larger communities.

These are all very breaking changes. Would it make sense to introduce them at the same time as the new compiler?

sbourdeauducq, 2021-05-11:
Thanks for the detailed analysis!
IMO we can easily switch to another protocol for IPC - note that this would apply to GUI/Applet communication as well.
Pickle has the advantage of being part of the standard Python library and not requiring any extra dependency. Is Dask worth the hassle?

sbourdeauducq, 2021-05-11:
> * Dask for serialization
> * mempack for header serialization

Did you mean msgpack?
Is there a simpler option? Two serialization libraries for this task sounds a bit heavy-handed.

> However, if we don't roll our own IPC and use `pyarrow` for all the functions above, that's not a problem and would simplify ARTIQ's codebase.

There are reasons for the IPC situation, in particular Windows and asyncio support. When I wrote this code, this wasn't available anywhere.

> Other candidates: I mentioned ZeroMQ earlier - this would be great for message routing, but doesn't come with any serialization, 

I did look into ZMQ also. It did not support asyncio, did not have all the features we wanted (for example, doing the initialization step in sync_struct is a pain with ZMQ publish/subscribe), and I don't see why a ZMQ socket would be faster than a regular Python socket unless we use ZMQ features for e.g. distributing messages to several clients. But this does not seem to be the bottleneck here, especially as the number of connected clients is typically small.

charlesbaynham, 2021-05-11:
PS PEP574 at https://www.python.org/dev/peps/pep-0574/ has some good insight into serialization, how it's currently done and how pickle is being improved in future.

sbourdeauducq, 2021-05-11:
And I'm aware of these features of ZeroMQ but they mostly do not seem relevant for our use cases. I doubt Python can really take advantage of the zero-copy API for instance, nor do I think that in an application that processes data in Python, a low-level memcpy in the socket layer can often be the bottleneck.

sbourdeauducq, 2021-05-11:
And portable ZeroMQ for IPC is TCP/IP-based, isn't it? Unix domain sockets for zmq_ipc do not work on Windows. So you're computing packet checksums and dealing with windows/acks for imaginary packet losses when sending data between processes, on a reliable medium. pipe_ipc in SiPyCo simply uses lightweight file descriptor inheritance on Unix, and IPC sockets on Windows.

charlesbaynham, 2021-05-11:
> 
> 
> Dask is also quite a heavy dependency and pulls a bunch of other ones (bokeh, pandas, SQLAlchemy, nodejs...)
> Can its serialization part be used separately?

Not sure. Dask itself pulls in the following:

```
    "pyyaml",
    "cloudpickle >= 1.1.1",
    "fsspec >= 0.6.0",
    "toolz >= 0.8.2",
    "partd >= 0.3.10",
```

and `dask.distributed` which is the part with the serialization uses

```
click >= 6.6
cloudpickle >= 1.5.0
dask >= 2021.03.0
msgpack >= 0.6.0
psutil >= 5.0
sortedcontainers !=2.0.0, !=2.0.1
tblib >= 1.6.0
toolz >= 0.8.2
tornado >= 5;python_version<'3.8'
tornado >= 6.0.3;python_version>='3.8'
zict >= 0.1.3
pyyaml
setuptools
```

So nothing too heavy there. `tornado` is probably the biggest. 

> ...nor do I think that in an application that processes data in Python, a low-level memcpy in the socket layer can often be the bottleneck.

You'd be surprised! When I was optimizing the Dask serialization I was able to trim 5ms off a previously 15ms deserialisation by being smarter about memoryview. 

> And portable ZeroMQ for IPC is TCP/IP-based, isn't it? Unix domain sockets for zmq_ipc do not work on Windows.

Yep I think you're right about that, reading into it a bit more. So probably not the right choice for IPC.

charlesbaynham, 2021-05-11:
Hmm I'll see how feasible that is. Not sure how coupled it is with the rest of the Dask code.