m-labs/artiq #1678: List elements not lifetime-tracked

pca006132, 2021-05-20:
# Bug Report

## One-Line Summary

Lifetime for list elements is not tracked, which can also cause memory corruption.

## Issue Details

List lifetime should also consider the lifetime of its elements. Otherwise, by assigning objects into a list and access them, we would not be able to track their lifetime.

Using an example similar to #1677:
```python
@kernel
def bar(ls):
    a = [1, 2, 3]
    b = [ls]
    b[0][0] = a
```

This should not compile, as the following equivalent code does not compile:
```python
@kernel
def bar(ls):
    a = [1, 2, 3]
    ls[0] = a
```

The expected output and actual output are similar to #1677.

### Your System (omit irrelevant parts)

* ARTIQ version: current master.

sbourdeauducq, 2021-10-03:
> This should not compile

NAC3 seems to be happy with is. Is it what you expected?

```python
@kernel
def bar(ls: list[list[int32]]):
    a = [1, 2, 3]
    b = [ls]
    b[0][0] = a

@kernel
def run() -> int32:
    bar([[2, 1], [1, 4]])
    return 0
```

pca006132, 2021-10-03:
> > This should not compile
> 
> NAC3 seems to be happy with it. Is it what you expected?
> 
> ```python
> @kernel
> def bar(ls: list[list[int32]]):
>     a = [1, 2, 3]
>     b = [ls]
>     b[0][0] = a
> 
> @kernel
> def run() -> int32:
>     bar([[2, 1], [1, 4]])
>     return 0
> ```

We have not yet implemented escape analysis in nac3.