m-labs/artiq #1679: MonInj for AD9910/AD9912

fanmingyu212, 2021-05-21:
# ARTIQ Feature Request

## Problem this request addresses

Set/get AD9910/AD9912 frequencies and amplitudes by monitor and injection. Currently, we need to run experiment(s) to set/read the values, and the preparation and compiling time for those experiments is a significant overhead (~1 s on a Windows machine).

## Describe the solution you'd like

Extend current MonInj (which I think only support AD9914) to support AD9910 and AD9912. This should allow operations such as manually changing the frequency / amplitude of DDSes with negligible delay.

If the limited bandwidth of the auxiliary connection to the device is a problem, maybe the user can select what channels they want MonInj to be on when they compile the firmware?

sbourdeauducq, 2021-05-21:
Injection is not practical. Monitoring can be done. The channel selection is already dynamic in the firmware and network protocol, what is missing is the GUI part. #676

fanmingyu212, 2021-05-21:
@sbourdeauducq I see. It is good news learning that monitoring can be done. Can you help me to understand why injection is not practical?

sbourdeauducq, 2021-05-21:
What happens if injection tries to access the SPI bus at the same time as the experiment?

fanmingyu212, 2021-05-21:
Can the injection be disabled when an experiment using Kasli is running? The injection can be active only when no experiment is running. I think users typically don't want to change DDSes manually when running an experiment anyways.

dhslichter, 2021-05-21:
@fanmingyu212 the moninj for the AD9914 is only monitoring, not injection, for what it's worth.  

I would argue that if you are just trying to manually play with DDS parameters, a better solution than changing moninj would be to make it possible to write an experiment that runs in an infinite loop and RPCs to check the value of a dataset on the PC (could be a DDS frequency or amplitude, a DAC voltage, anything).  Then if the user changes the value of the dataset manually, this is broadcast to the master and then the next time the experiment RPCs to check the dataset value, it gets the updated value and then does whatever the experiment code wants it to do.  Because this is very general, and the desired tasks are on human (aka slow) timescales, the latency involved with this RPCing wouldn't be a problem.  

My understanding is that this won't work right now because the manual changing of the datasets doesn't broadcast in the needed way?  But I haven't tried it myself to see.

sbourdeauducq, 2021-05-21:
The manual dataset changes are also broadcast (if you use the right options), and that would work right now.

fanmingyu212, 2021-05-21:
@dhslichter thanks for the suggestion. We can definitely write an experiment running an infinite loop and use RPCs to check values, but I think MonInj could be an easier method for ARTIQ users if it is implemented (I don't know how much work it would involve to implement MonInj on DDSes though).

For our use case, we have a background PMT counting experiments that usually runs on a loop of ~100-200 ms (which is the PMT count time), but the loop time is occasionally changed to ~1 s (_e.g._, during check clock laser alignment). It is probably natural to check for DDS changes after each loop, but a ~1 s delay time when changing a DDS parameter is a bit long when we are actively tuning parameters. We can implement inner loops checking for DDS parameter changes to allow for faster update rates, but it seems like a convoluted way to solve this problem from a user perspective. Instead, if there is a MonInj protocol it could reduce much complexity in this background experiment.

dhslichter, 2021-05-21:
I think that the point is that setting up moninj to inject DDS parameters is a very challenging problem in general, and also requires a lot of gateware adjustment, while the other solution can be carried out entirely in software.  I would say that monitoring (as opposed to injection) might be a nice feature for the DDS parameters, and would be less disruptive although there are potential issues if you start to have many channels (see #676).

jordens, 2021-05-21:
Duplicate of #1142

dtcallcock, 2021-05-21:
These kind of UI improvement suggestions keep popping up but nothing has really moved on the dashboard for a few years/versions now.  What would it take to get the momentum to do a decent refresh on it?

hartytp, 2021-05-21:
> These kind of UI improvement suggestions keep popping up but nothing has really moved on the dashboard for a few years/versions now. What would it take to get the momentum to do a decent refresh on it?

Funding/someone to drive it AFAICT.

I still think that a lot of the discussion around https://github.com/m-labs/artiq/issues/676 stands. e.g. rather than building specific widgets for the DDSs/DACs into ARTIQ, it would make more sense to treat moninj as another data source with the master mediating between clients and the FPGA (sipyco/MQTT/whatever). e.g. I'd love to be able to write a noise-eater configuration applet to interface with SU-servo, or hook my servos up to telegraf for logging.

There's also a bit of a refactoring of the gateware moninj level (e.g. adding an address to the bus rather than relying on data being available all the time)

hartytp, 2021-05-21:
see also all the related issues that make sense to sort out in an overhaul like https://github.com/m-labs/artiq/issues/1435

sbourdeauducq, 2021-06-15:
Closing, see #1142 instead.

closed 2021-06-15 by sbourdeauducq