m-labs/artiq #1684: [docs] Type error on delay_mu(self.core.ref_multiplier)

airwoodix, 2021-05-27:
# Bug Report

## Summary

Since #1461, the manual suggests to use `delay_mu(self.core.ref_multiplier)` as an option to avoid sequence errors. On b89610bb this fails because `core.ref_multiplier` is implicitly a `int32` and `delay_mu()` expects a `int64`.

What's better?
  - update the manual to recommend `delay_mu(np.int64(self.core.ref_multiplier))`;
  - cast `ref_multiplier` to `int64` in `artiq.coredevice.core.Core.__init__`;
  - introduce a function, e.g. `delay_rtio_coarse()`, if this is seen as a (very) common use case.

### Your System

* ARTIQ version: 7.7634.b89610bb.beta

lriesebos, 2021-05-27:
We also run periodically into this behavior. It is expected behavior, but it is not convenient.

In our code, we only use `ref_multiplier` in the context of time in machine units. Casting the value to `int64` in `artiq.coredevice.core.Core.__init__` would probably be a convenient solution for us. Though in the end, I think any of the proposed solutions is fine with us.

sbourdeauducq, 2021-05-27:
There's no good reason for ref_multiplier to be int64. It's always a small number.

>     * if this is seen as a (very) common use case.

Is it?

lriesebos, 2021-05-28:
Specifically for us, I would not call it a very common use case, but it is our only use case for this variable at this moment. A quick search through the ARTIQ repo also suggests that this variable is mainly used in the context of time in machine units.

I think adding `delay_rtio_coarse()` is overkill, but any of the other solutions would work for us.

airwoodix, 2021-05-28:
> There's no good reason for ref_multiplier to be int64. It's always a small number.

Agreed. I was mostly reacting to the fact that the example in the manual doesn't work.

> Is it?

On user code, I don't know. 
In the coredevice drivers, there're a couple of isolated hits. For example `spi2` and `suservo` do 

```python
self.ref_period_mu = self.core.seconds_to_mu(
    self.core.coarse_ref_period)
assert self.ref_period_mu == self.core.ref_multiplier
# (...)
delay_mu(self.ref_period_mu)
```
and make use of `ref_period_mu` outside of `delay_mu()` calls.

`grabber` and `sawg` on the other hand locally cast `core.ref_multiplier` (named `c` and `coarse_cycle` respectively).

Instead of casting `core.ref_multiplier`, would it make sense to expose `ref_period_mu` in the `core` instance (maybe as `coarse_ref_period_mu` for consistency) such that `delay_mu(self.core.ref_period_mu)` works?

PR #1701 (2021-06-23): docs: fix RTIO coarse delay snippet and misc. formatting

# ARTIQ Pull Request

## Description of Changes

- Fixes #1684 by using ``delay`` instead of ``delay_mu``, since ``coarse_ref_period`` is already available in the core instances. Exposing ``coarse_ref_period_mu = core.seconds_to_mu(coarse_ref_period)`` would also be an option but results in more verbose user code. For consistency, one would also need to update an number of coredevice drivers (``spi2``, ``grabber``, ``sawg``, ``suservo``) which is quite intrusive.
- Fix miscellaneous formatting issues in the docs.

## Type of Changes

|  [... cut]
closed 2021-06-23 by sbourdeauducq