m-labs/artiq #1685: A FAQ would be nice

jayich, 2021-05-28:
A FAQ that the community could contribute to and keep updated would probably help reduce the learning curve's slope.  Possibly it would be good to have it editable via a git repo so that the git toolset may be utilized.  It would be great to hear from others if this is desirable, and how they think this might be implemented effectively.

dhslichter, 2021-05-28:
I think this could be helpful, particularly for common questions for new users.  It might also be a good place to give some basic information on things like how to make issue posts most productive (e.g. avoiding XY problems, searching past issues or manual, etc), general style and design philosophy for ARTIQ (i.e. aiming for generality and flexibility, trying not to load up with a ton of specific features that are only relevant for a few users, attacking large design problems in stages rather than trying to spec/do the whole thing at once, etc).  

A few thoughts and questions:
- This seems like something that would need to be curated and managed by some experts so that it is of high quality and doesn't rot.  We'd have to get buy-in from some appropriate people.
- Where should/would this be hosted?  On this git repo, or a separate one?  Could the compiled version appear as a section of the manual, for example?
- To what extent should it include Sinara-specific FAQ items?

dhslichter, 2021-05-28:
My own opinions on the above:

- I would suggest that this FAQ be curated by community members (experienced users and/or developers) and not just M-Labs.  For that reason, I think it might be good to have its own separate repo so we can give the appropriate people permissions to manage it.  It might still be under the m-labs organization, though.  
- I think we should focus on ARTIQ-related stuff, but obviously some of it is inseparable from Sinara.  It might be good to have a similar FAQ about Sinara on the sinara-hw organization somewhere, and have them link to each other?  I would divide things as hardware-only belonging in the Sinara FAQ, but anything that touches Sinara gateware belonging in the ARTIQ FAQ?

aktentasche, 2021-06-01:
I'd be willing to support and contribute and I think it is a great idea. In the end Sinara and ARTIQ are hard to separate I think. It also depends on what M-labs thinks of this.

airwoodix, 2021-06-02:
@HarryMakes Could such a FAQ have some overlap with this project? https://git.m-labs.hk/harry/mlabs-kb-cobalt

I think that a wiki would be worth considering as an alternative to a knowledge base. I see that the KB, coupled with VCS/PR, provides authoritative, peer-reviewed information, but the contribution step is high, as can be seen from the relatively few contributions to the manual in this repository. IMO, if users are ready to take the information presented on a wiki with a little grain of salt, examples like the [ArchWiki](https://wiki.archlinux.org) are very convincing to provide up-to-date information with fast user feedback.

While certain groups certainly use ARTIQ without Sinara, the vast majority surely makes the projects synonymous (as-in "the ARTIQ DDS modules"). I think that the gap to fill is documentation on how to use ARTIQ with Sinara (and vice-versa). At the moment, if I want full documentation on running e.g. Urukul with ARTIQ, I need to visit three different repositories in three different organizations: [sinara-hw/Urukul](https://github.com/sinara-hw/Urukul) (the HW design), [quartiq/urukul](https://github.com/quartiq/urukul) (the PL design), [m-labs/artiq](https://github.com/m-labs/artiq) or m-labs.hk (the coredevice drivers). It certainly makes sense to keep these repositories separated (I guess also for documentation maintainability reasons?) but this is missing overview.

Would it be an option to build on the Sinara wikis and add ARTIQ-specific information there? The Sinara hardware can be used without ARTIQ, but ARTIQ is the main entry-point to Sinara hardware (or vice-versa?), except maybe for some Humpback/Banker-based designs.

dtcallcock, 2021-06-03:
>Would it be an option to build on the Sinara wikis and add ARTIQ-specific information there?

It could be for sure.  Just be aware though that the GitHub wikis have pretty limited functionality.  For the Sinara board wikis it's likely worth living with to have the wikis nicely integrated into the repositories.

On the other hand setting up something stand-alone adds like the 9th place I need to check if I have an issue...

dtcallcock, 2021-06-03:
> On the other hand setting up something stand-alone adds like the 9th place I need to check if I have an issue...

On the other hand, it could improve the problem if it replaced and consolidated some of the static-ish website jumping-off points we currently have (eg. https://m-labs.hk/experiment-control/resources/ and https://sinara-hw.github.io).  There would be one url we could send all people to where they could get a picture of everything. Would be good to get @sbourdeauducq take here.

sbourdeauducq, 2021-06-03:
I don't think the "resources" page can be replaced completely, or at least not its "community code" section.
I've been wanting to make Sinara datasheets for a while (collecting all information scattered around and taking new measurements), but I've been too occupied by other things. For the hardware itself, I'm not sure if a wiki makes sense, since it's pretty static. There's already a (not very good) FAQ in the ARTIQ manual: https://m-labs.hk/artiq/manual/faq.html
What about sending PRs against it which we could then review? The FAQ could have subsections dedicated to each Sinara card.

aktentasche, 2021-06-04:
I think sending PRs is a good idea, however then the overall curation is missing. Also I doubt that a lot of people will think "ah now I have an answer to my question, let me create a PR".

An option could be to curate the input to the FAQ first and then send a PR, with the already mentioned community experts being the curators. Initially this team could go through github issues and forum posts to extract which questions are actually frequently asked.

architeuthidae, 2024-08-14:
Summarizing this discussion a few years later: 
1. We would like a FAQ which covers (a) common problems for new users, where to find other resources, how to write productive issues, actually frequently asked questions asked in the forum and helpdesk (b) the general philosophy of ARTIQ, basic use of Sinara (possibly in the spirit of #1354)
2. It should be curated by experts (though perhaps community experts and not just M-Labs) and editable with the Git toolset 
3. There is some uncertainty on whether it should be connected to the Sinara wikis, the ARTIQ manual, a third unrelated location, etc. 
4. General consensus is there are tendentially already too many places to check to solve problems (all of which are of course variously out of date in practice) and adding another doesn't necessary make things more accessible, let alone easier to maintain (cf [XKCD](https://xkcd.com/927/)) 

As things stand after recent rewrites, I think there's a decent case to be made [the manual FAQ](https://m-labs.hk/artiq/manual/faq.html) is pretty much the answer here. Topics in (a) are covered, (b) could be added at any time, and the 'Additional Resources' tab also attempts to be an up-to-date directory for everywhere good information can be found, interlinked both ways with the m-labs.hk resources page. Perfectly centralized documentation definitely isn't really possible here, so IMO, the next best option is keeping everything well interlinked, so no matter where you start you bump into all the other possible resources early (and right now, the ideal place to start probably _is_ the manual, which is the most centralized and, for now at least, up-to-date, of the available options.) It is admittedly only curated by M-Labs, but this also has its advantages, and it does now go out of its way to encourage additions or corrections, so hopefully PRs indeed can and will be sent. 

If there's no objections, this probably qualifies as a "done."