m-labs/artiq #1687: Conda install fails for python>=3.8 on windows

ljstephenson, 2021-05-31:
# Bug Report

## One-Line Summary

Conda install fails for Python versions 3.8 and higher.

## Issue Details

I know we all love to bash conda's dependency "solver": conda install fails because of package conflicts, which it really shouldn't.

### Steps to Reproduce

```
(base)>type .condarc
channels:
  - https://conda.m-labs.hk/artiq
  - defaults
  - conda-forge

(base)>conda create -n "trial" python=3.9 artiq
```

### Expected Behavior

Should create a conda environment with python 3.9 and artiq...

### Actual (undesired) Behavior

* Conda's dependency solver gets confused and lists conflicts.
* Note that in the below output, `h5py` seems to be involved in the dependency mess, but `conda create -n "trial" python=3.9 h5py` succeeds
* Tried on multiple machines and with fresh anaconda installl
* Command output:
```
(base)>conda create -n "trial" python=3.9 artiq
Collecting package metadata (current_repodata.json): done
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: |
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
                                                                                                                       |
failed

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versions

Package python conflicts for:
python=3.9
artiq -> h5py -> python[version='2.7.*|3.5.*|3.6.*|>=2.7,<2.8.0a0|>=3.6,<3.7.0a0|>=3.7,<3.8.0a0|>=3.9,<3.10.0a0|>=3.8,<3.9.0a0|>=3.5,<3.6.0a0|3.4.*|<3.8|>=3.5|>=3.7|>=3.6']
```

### Your System (omit irrelevant parts)

* Operating System: W10
* Conda: 4.10.1

sbourdeauducq, 2021-06-01:
Besides conda problems, binary packages that ARTIQ uses need to be recompiled against Python 3.9.
Python 3.9 is not supported with conda at the moment.

ljstephenson, 2021-07-02:
This also fails in exactly the same way for Python 3.8; is there a timeline to get the binaries compiled against Python>3.7?

sbourdeauducq, 2021-07-05:
If you want to help, find out on Linux why the installer from ``Anaconda3-2021.05-Linux-x86_64.sh`` fails with ``error while loading shared libraries: libz.so.1: cannot open shared object file: No such file or directory`` in the FHS environment, even though ``libz.so.1`` is present in ``/usr/lib`` and ``/lib``.
On Windoze there seems to be no problems with recent conda, but we need Linux for noarch packages and non-Nix users.

sbourdeauducq, 2021-07-05:
Done for ARTIQ 6 and 7, they now install with Python 3.8 and the environment that comes with the latest Anaconda (2021.05). Note: conda (and windoze) is still a pile of junk, the only reason I could do such a major update in a few hours is thanks to WFVM and related infrastructure.

closed 2021-07-05 by sbourdeauducq