m-labs/artiq #1688: sinara_tester/mirny: channel setup underflows

airwoodix, 2021-06-02:
# Bug Report

## One-Line Summary

`artiq_sinara_tester` underflows in [setup_mirny()](https://github.com/m-labs/artiq/blob/2671c271d4e9d18d48b3e424912f0c30fa0d9aa3/artiq/frontend/artiq_sinara_tester.py#L296-L306).

## Issue Details

I found the following implementations to remove the RTIO underflow:

```python
@kernel
def setup_mirny(self, channel, frequency):
    self.core.break_realtime()
    delay(1*ms)  # NEW
    channel.init()
    channel.set_att_mu(160)
    channel.sw.on()
    channel.set_frequency(frequency*MHz)
```

```python
@kernel
def setup_mirny(self, channel, frequency):
    at_mu(self.core.get_rtio_counter_mu() + 3*125000)
    channel.init()
    # etc.
```

  - @occheung can you reproduce this issue?
  - what's the real problem here? Does the `print()` RPC just take too long to return on my system? (testing Urukuls or Samplers works)
  - the extra delays in the [current implementation](https://github.com/m-labs/artiq/blob/2671c271d4e9d18d48b3e424912f0c30fa0d9aa3/artiq/frontend/artiq_sinara_tester.py#L296-L306) are not necessary.

### Your System

* Operating System: Linux / nix
* ARTIQ version: v7.7634.b89610bb.beta
* Version of the gateware and runtime loaded in the core device: same as host
* Hardware involved: Mirny v1.1, Kasli v1.1

occheung, 2021-07-06:
Repeatable in ARTIQ `7.7663.51d2861e` with iIdentical hardware spec.

Here are some log from `artiq_coreanalyzer -p` after running `artiq_sinara_tester`.
```
Log channel: 7
DDS one-hot: True
OutputMessage(channel=5, timestamp=1057491096648, rtio_counter=1057490975016, address=0, data=1)
OutputMessage(channel=5, timestamp=1057591096648, rtio_counter=1057490976824, address=0, data=0)
OutputMessage(channel=5, timestamp=1057691096648, rtio_counter=1057490977712, address=0, data=1)
OutputMessage(channel=5, timestamp=1057791096648, rtio_counter=1057490978136, address=0, data=0)
OutputMessage(channel=5, timestamp=1057891096648, rtio_counter=1057490978616, address=0, data=1)
OutputMessage(channel=5, timestamp=1057991096648, rtio_counter=1057490979040, address=0, data=0)
OutputMessage(channel=5, timestamp=1058091096648, rtio_counter=1057891097816, address=0, data=1)
OutputMessage(channel=5, timestamp=1058191096648, rtio_counter=1057891098888, address=0, data=0)
OutputMessage(channel=5, timestamp=1058291096648, rtio_counter=1057891099360, address=0, data=1)
OutputMessage(channel=5, timestamp=1058391096648, rtio_counter=1057891099848, address=0, data=0)
OutputMessage(channel=5, timestamp=1058491096648, rtio_counter=1057891100312, address=0, data=1)
OutputMessage(channel=5, timestamp=1058591096648, rtio_counter=1057891100904, address=0, data=0)
OutputMessage(channel=5, timestamp=1058691096648, rtio_counter=1058491097592, address=0, data=1)
OutputMessage(channel=5, timestamp=1058791096648, rtio_counter=1058491098688, address=0, data=0)
OutputMessage(channel=5, timestamp=1058891096648, rtio_counter=1058491099168, address=0, data=1)
OutputMessage(channel=5, timestamp=1058991096648, rtio_counter=1058491099648, address=0, data=0)
OutputMessage(channel=5, timestamp=1059091096648, rtio_counter=1058491100128, address=0, data=1)
OutputMessage(channel=5, timestamp=1059191096648, rtio_counter=1058491100552, address=0, data=0)
OutputMessage(channel=5, timestamp=1059291096648, rtio_counter=1059091097320, address=0, data=1)
OutputMessage(channel=5, timestamp=1059391096648, rtio_counter=1059091098480, address=0, data=0)
OutputMessage(channel=5, timestamp=1059491096648, rtio_counter=1059091098944, address=0, data=1)
OutputMessage(channel=5, timestamp=1059591096648, rtio_counter=1059091099376, address=0, data=0)
OutputMessage(channel=5, timestamp=1059691096648, rtio_counter=1059091099968, address=0, data=1)
OutputMessage(channel=5, timestamp=1059791096648, rtio_counter=1059091100392, address=0, data=0)
OutputMessage(channel=6, timestamp=1059891096648, rtio_counter=1059691098368, address=0, data=1)
OutputMessage(channel=6, timestamp=1059991096648, rtio_counter=1059691100144, address=0, data=0)
OutputMessage(channel=6, timestamp=1060091096648, rtio_counter=1059691101368, address=0, data=1)
OutputMessage(channel=6, timestamp=1060191096648, rtio_counter=1059691101888, address=0, data=0)
OutputMessage(channel=6, timestamp=1060291096648, rtio_counter=1059691102360, address=0, data=1)
OutputMessage(channel=6, timestamp=1060391096648, rtio_counter=1059691102792, address=0, data=0)
OutputMessage(channel=0, timestamp=1060491096648, rtio_counter=1059852998712, address=1, data=17700622)
OutputMessage(channel=0, timestamp=1060491096656, rtio_counter=1059853005320, address=0, data=0)
InputMessage(channel=0, timestamp=0, rtio_counter=1060491099808, data=256)
OutputMessage(channel=0, timestamp=1060491199864, rtio_counter=1060491103384, address=1, data=16914186)
OutputMessage(channel=0, timestamp=1060491199872, rtio_counter=1060491106784, address=0, data=50331648)
OutputMessage(channel=0, timestamp=1061296585336, rtio_counter=1061296701608, address=1, data=16910088)
ExceptionMessage(channel=0, rtio_counter=1061296701624, exception_type=<ExceptionType.o_underflow: 20>)
StoppedMessage(rtio_counter=1065391755536)
```
It might be useful to map the `data` to the corresponding transactions that was queued into the FIFO.

And here is a snippet of my `device_db.py`.
```python
device_db["spi_mirny0"]={
    "type": "local",
    "module": "artiq.coredevice.spi2",
    "class": "SPIMaster",
    "arguments": {"channel": 0x000000}
}

# omitted

device_db["led0"] = {
    "type": "local",
    "module": "artiq.coredevice.ttl",
    "class": "TTLOut",
    "arguments": {"channel": 0x000005}
}

device_db["led1"] = {
    "type": "local",
    "module": "artiq.coredevice.ttl",
    "class": "TTLOut",
    "arguments": {"channel": 0x000006}
}
```

occheung, 2021-07-06:
>what's the real problem here? Does the print() RPC just take too long to return on my system? (testing Urukuls or Samplers works)

Given that there is a `self.core.break_realtime` invocation, I don't think the `print` in `test_mirnies` matters at all. 

I put some prints here to reduce formatting time loss.

```python
    def setup_mirny(self, channel, frequency):
        
        print(self.core.get_rtio_counter_mu())
        print(now_mu())

        self.core.break_realtime()

        print(self.core.get_rtio_counter_mu())
        print(now_mu())

        channel.init()
```

```
5114969759944    # Original RTIO counter
5113895111768    # Original timeline
5114974182440    # RTIO counter after break_realtime
5114974305864    # Timeline after break_realtime
```

It seems that `now_mu() > get_rtio_counter_mu()`, but it might not be large enough. If I am not mistaken, the following output event (see previous comment):
```
OutputMessage(channel=0, timestamp=1061296585336, rtio_counter=1061296701608, address=1, data=16910088)
```
is the first event in `adf5356.init()`, and it cannot even queue in.
(Needless to say it definitely should still cause RTIO underflow)

occheung, 2021-07-07:
One source of large time loss can be found in `adf5356.sync()`.
```python
    @kernel
    def sync(self):
        """
        Write all registers to the device. Attempts to lock the PLL.
        """
        pre_pfd_counter = self.core.get_rtio_counter_mu()
        f_pfd = self.f_pfd()
        post_pfd_counter = self.core.get_rtio_counter_mu()
        print(pre_pfd_counter)
        print(post_pfd_counter)

        # omitted
```
Which gives the following output.
```
69426638786176
69426638888104
```
A quick subtraction would see that `adf5356.f_pfd()` took ~100000 machine unit.

Another source that takes a lot of time is the calculation of PLL, also in `adf5356.sync()`.
```python
    @kernel
    def sync(self):
        """
        Write all registers to the device. Attempts to lock the PLL.
        """
        f_pfd = self.f_pfd()

        if f_pfd <= 75.0 * MHz:
            for i in range(13, 0, -1):
                self.write(self.regs[i])
            delay(200 * us)
            self.write(self.regs[0] | ADF5356_REG0_AUTOCAL(1))
        else:
            # AUTOCAL AT HALF PFD FREQUENCY

            # calculate PLL at f_pfd/2
            pre_pll_counter = self.core.get_rtio_counter_mu()
            n, frac1, (frac2_msb, frac2_lsb), (mod2_msb, mod2_lsb) = calculate_pll(
                self.f_vco(), f_pfd >> 1
            )
            post_pll_counter = self.core.get_rtio_counter_mu()
            print(pre_pll_counter)
            print(post_pll_counter)

            #omitted
```
Which gives the output:
```
69973804527504
69973804660832
```
That would be another >100000 machine unit.
We only get 125000 from `core.break_realtime()`. If the whole `adf5356.init()` method is not intended to work without manually adding `delay()` in experiment scripts, then we should modify `artiq_sinara_tester.py` as you suggested. Otherwise, we probably should modify the delays inside `adf5356.sync()`.

sbourdeauducq, 2021-07-07:
In my opinion this is what should be done:
1. On master, fix the ``sync()`` function by adding appropriate delays. Mention in the release notes that these delays have been changed.
2. On release-6, add a delay in artiq_sinara_tester before ``channel.init()`` to stop the underflow.

airwoodix, 2021-07-07:
Thanks for looking into this.

> If the whole ``adf5356.init()`` method is not intended to work without manually adding ``delay()`` in experiment scripts, then we should modify ``artiq_sinara_tester.py`` as you suggested.

The experiment below works has no extra delays and doesn't underflow. What's the difference to ``artiq_sinara_tester``?

```python
from artiq.experiment import *
from pprint import pprint
from numpy import int64

class Mirny(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.mw = self.get_device("mirny0_ch0")

        self.pre = int64(0)
        self.port = int64(0)

    @kernel
    def run(self):
        self.core.reset()
        self.mw.cpld.init()
        self.mw.init()
        self.mw.set_att_mu(160)
        self.mw.sw.on()
        self.mw.set_frequency(800*MHz)

        self.pre = self.core.get_rtio_counter_mu()
        _ = self.mw.f_pfd()
        self.post = self.core.get_rtio_counter_mu()

    def analyze(self):
        pprint(self.mw.info())
        print(self.post - self.pre)
```

occheung, 2021-07-07:
IIRC `mirny.init()` gives a very large delay (1 ms) at the end.
https://github.com/m-labs/artiq/blob/b10d1bdd378a9a9f904b97c1434c36fcc619f112/artiq/coredevice/mirny.py#L122-L123
We can expect a much larger difference between `rtio_counter` and `now` than 125000, calling `mirny.init()`  is probably a free delay.