m-labs/artiq #1691: save custom attributes in archived h5py data files

fanmingyu212, 2021-06-04:
# ARTIQ Feature Request

## Problem this request addresses

Currently users can only save HDF5 datasets in the `datasets` group of a file. This does not provide a clean way to save custom metadata, which include parameters not fetched as experiment arguments and other experiment parameters (wavemeter frequencies, ion trap voltages, etc.).

## Describe the solution you'd like

Some solutions are:
1. Provide access to [HDF5 attributes](https://docs.h5py.org/en/stable/high/attr.html) of datasets and groups
2. Provide functions to create new groups, and save custom attributes as datasets in custom groups.

Comments/ideas from others would be appreciated. It would be helpful to know how the more advanced ARTIQ users save their metadata in the data file.

airwoodix, 2021-06-04:
This is linked to #1545 and references therein.

wlph17, 2022-10-12:
about the HDF5 attributes, which one is better, setting them one by one or setting them all by passing as a dict in one function?

fanmingyu212, 2023-06-14:
> Why not just save those as output datasets? Taking laser wavelength and ion trap voltage measurements is pretty much part of a regular Experiment.

I would prefer to save these variables in a different hdf5 group than "datasets" to organize data better. We also save data such as AOM frequency shift direction, single/double pass, etc. These configurations seem more like attributes than datasets to me, though I agree we could save them as datasets too.

Currently we use a homebuilt server to save hdf5 files in custom format, so I can close this issue if you would like.