m-labs/artiq #1693: larger RTIO input buffer?

fanmingyu212, 2021-06-07:
# Question

Would it be possible to make kasli be able to buffer more input timestamps?

## Category: development/documentation

## Description

When using functions such as `gate_rising_mu` and `timestamp_mu` to collect and readout timestamps, only a small number of RTIO inputs (64?) can be buffered by kasli before it raises a `RTIOOverflow` exception. Would it be possible to increase the number of timestamps it can buffer to a significantly larger number such as 1024?

sbourdeauducq, 2021-06-07:
Yes, with a corresponding increase in FPGA resource usage.

sbourdeauducq, 2021-06-07:
We can do it on individual builds or channels but I would not increase the default significantly.

closed 2021-06-07 by sbourdeauducq

fanmingyu212, 2021-06-07:
Could it be good to add a guide to increase this number in the documentation or in this issue? I assume the firmware needs to be rebuilt for this change, and the increased usage should be fine as long as the firmware building succeed. Therefore, it could be good to allow groups operating on self-built firmware to increase this buffer.

dhslichter, 2021-06-07:
@fanmingyu212 we have done this for custom gateware builds for some of our devices.  As @sbourdeauducq points out, there are limited FPGA resources, so generally doing this as a blanket "deep" queue for all channels won't win you much.  However, if you know there are a few specific channels where you want really deep queues, and then the rest can be the default of 64, then you can probably make those few "deep" queues as deep as 1024 if you like.  You have to try building gateware using the Xilinx tools to find out.  

And yes, you will need to build firmware from source.  The code for an RTIO channel can take an argument to change the `ififo_depth`, which will give you different depths for the input queues.  See here:
https://github.com/m-labs/artiq/blob/master/artiq/gateware/rtio/channel.py

You adjust this in the appropriate gateware target file.  If you are using a KC705, this can be defined by adjusting things in the file here: https://github.com/m-labs/artiq/blob/master/artiq/gateware/targets/kc705.py .  For Kasli, things are factored into lower levels, but it can still be done.  Basically, you would need to insert some calls when building up the `target` to put in some DIO channels with the increased FIFO depth, which could be done using the same kind of code as for the KC705, but you need to be careful as the Kasli DIO code is currently configured to do things automatically for one EEM at a time, and it's not immediately clear to me that one can do this without either hacking the code of the `DIO.add_std()` method in https://github.com/m-labs/artiq/blob/master/artiq/gateware/eem.py to enable `ififo_depth` parameters to be passed to the `target.rtio_channels.append(rtio.Channel.from_phy(phy))` as an additional argument to the innermost function, or by manually populating all of the RTIO channels for a given EEM on which you will have the DIOs with custom input FIFO depths.  

Anyway, this doesn't count as documentation, but hopefully helps you understand what you would need to do to accomplish your goals.  IMHO this would be best documented in two places: one would be the "ARTIQ real-time I/O concepts" section, where you basically tell users that there are finite depths to the ififo queues, usually 64 events (512 for the NIST gateware on KC705, because it's a bigger FPGA with fewer digital I/O channels competing for resources).  The second thing would be to have a "superusers: building your own gateware" section.  Right now, the "Developing ARTIQ" is not super detailed.  

Linking to #1685 -- this is a great FAQ question/answer.

fanmingyu212, 2021-06-08:
@dhslichter thank you for the detailed answer to my question!

> For Kasli, things are factored into lower levels of code, but it can still be done...

This is extremely helpful for me. I will test this when I have time, and post steps/results here.

> 512 for the NIST gateware on KC705

This is encouraging for us. I just looked at the new artiq_zynq and it seems to have RTIO input buffer lengths of 512 as well: https://git.m-labs.hk/M-Labs/artiq-zynq/src/branch/master/src/gateware/zc706.py#L229. The larger buffer would help us accelerate some experiments that record PMT timestamps.

> IMHO this would be best documented in two places: one would be the "ARTIQ real-time I/O concepts" section, where you basically tell users that there are finite depths to the ififo queues, usually 64 [...]

I definitely agree that this information should be in the artiq manual or somewhere easy to access/search. Currently I found a lot of helpful information scattered in the GitHub issues and the M-labs forum but not gathered in a document. This will definitely help new users of ARTIQ like me.

dnadlinger, 2021-06-08:
IIRC you can just set `ififo_depth` after the fact, although I'm not using that myself:
```py
        eem.DIO.add_std(self, 0,
                        ttl_serdes_7series.InOut_8X, ttl_serdes_7series.Output_8X,
                        edge_counter_cls=SimpleEdgeCounter)
        self.rtio_channels[0].ififo_depth = 1024  # Deep ififo for PMT
```

dhslichter, 2021-06-08:
good call @dnadlinger -- too busy think about methods to just modify the object directly 😛

sbourdeauducq, 2021-06-09:
Hmm, this may only work because of some implementation details in Migen and ARTIQ, and the new value may not be used consistently in the gateware, leading to bugs. I would not recommend doing this.

dhslichter, 2021-06-09:
@sbourdeauducq so would it be reliable to modify the `eem.DIO.add_std()` method to enable ``**kwargs` of some sort to be passed down to the `target.rtio_channels.append(rtio.Channel.from_phy(phy))` call?  If so, this might be something nice to upstream?

sbourdeauducq, 2021-06-09:
Yes.

dhslichter, 2021-06-09:
This code is already quite complex, and the layers of abstraction do make it challenging for less experienced users who want to create their own bitstreams.  I think that building custom bitstreams is something I would expect a fair number of groups might be interested in, even if they have no other interest in the inner workings of ARTIQ.  Some documentation on how this works would not be a bad idea, we may write something up and submit a PR.  

I think the DIO ififo depth is where this issue is most relevant; for ofifo one can typically use DMA instead (although not for all cases), and for the various other EEM cards, I think the default values are probably generally fine.  Would it be nicer to add an `eem.DIO.add_custom()` method which gives this more fine grained control on a per-channel basis, specifying the TTL class for each of the 8 channels (say as a list) with optional arguments for `ififo_depth` and `ofifo_depth` also?

hartytp, 2021-06-09:
> IIRC you can just set ififo_depth after the fact, although I'm not using that myself:

Indeed. This is exactly what I tend to do.

> Hmm, this may only work because of some implementation details in Migen and ARTIQ, and the new value may not be used consistently in the gateware, leading to bugs. I would not recommend doing this.

At least at present (from a quick search of the code), the FIFO depth is not used until the RTIO core is created. So, any modifications to the FIFO depth are totally fine until `add_rtio` is called. So, at least with the way things are currently structured, this seems fine to me

dhslichter, 2021-06-09:
> At least at present (from a quick search of the code), the FIFO depth is not used until the RTIO core is created. So, any modifications to the FIFO depth are totally fine until add_rtio is called. So, at least with the way things are currently structured, this seems fine to me

Even if this is true, I would contend that for code clarity reasons it probably would make sense to just write a new method that allows one to define a fully-customizable set of DIO channels.  This would make things much more obvious/apparent for newcomers to the code who are looking to do this kind of very standard thing (customizing their DIOs on a channel-by-channel basis).

han94ros, 2021-08-07:
> IIRC you can just set `ififo_depth` after the fact, although I'm not using that myself:
> 
> ```python
>         eem.DIO.add_std(self, 0,
>                         ttl_serdes_7series.InOut_8X, ttl_serdes_7series.Output_8X,
>                         edge_counter_cls=SimpleEdgeCounter)
>         self.rtio_channels[0].ififo_depth = 1024  # Deep ififo for PMT
> ```

I have implemented this change in Kasli.py however I'm not sure this change is being recognised as changing the depth from 64 to 1024 does not solve the RTIO overflow error, but more importantly decreasing the depth to 1 does not give rise to RTIO overflow errors on sequence runs which would run normally with an ififo depth of 64.

fanmingyu212, 2021-08-07:
@han94ros you might want to try the code in this PR: https://github.com/m-labs/artiq/pull/1696

han94ros, 2021-08-07:
> @han94ros you might want to try the code in this PR: #1696

I am looking to increase the ififo depth of the spi_sampler0_adc channel which is channel 20 in my device database. I assume this is a TTLinOut channel. Will this fix work on this channel?

fanmingyu212, 2021-08-07:
I see. I don't think the fix will work for a sampler channel.

han94ros, 2021-08-07:
> I see. I don't think the fix will work for a sampler channel.

This is an SPI2 master channel, as seen below. 

`device_db["spi_sampler0_adc"] = {
    "type": "local",
    "module": "artiq.coredevice.spi2",
    "class": "SPIMaster",
    "arguments": {"channel": 0x000014}`

I thought this could be treated the same as standard TTL channels but the TTL channels on the sampler are defined clearly. 

`device_db["ttl_sampler0_cnv"] = {
    "type": "local",
    "module": "artiq.coredevice.ttl",
    "class": "TTLOut",
    "arguments": {"channel": 0x000016},`

Do you have any ideas on how I could increase the ififo depth for the SPI2 Master channel