m-labs/artiq #1697: [RFC] add channel names to RTIO errors

hartytp, 2021-06-11:
At present RTIO errors (underflow, overflow, sequence) give only a channel number. That can make debugging experiment code quite painful.

It would be really nice if we provided more human-understandable channel descriptions as well as/instead of the channel numbers. Maybe grab a device name from the device db and then (as a luxury) add some driver-dependent additional information (something like `dds0.spi`).

dnadlinger, 2021-06-11:
+1; this is a surprisingly obvious idea that I can't believe didn't occur to me earlier.

dnadlinger, 2021-06-11:
(This would be especially useful in situations where you don't have a backtrace available, which often provides enough context otherwise.)

hartytp, 2021-06-11:
okay, thanks for the input @dnadlinger It felt so obvious that I wasn't sure if there was a reason it's actually completely infeasible.

Okay, I'll add that to our to fix list...

dnadlinger, 2021-06-11:
Obvious as in a feature that should obviously exist. :) The implementation might require some finesse, though, in terms of finding a way that avoids breaking the various abstractions as much as possible. Presumably the artiq.coredevice.core code would need to inspect the exception types, and on a match interpret the extra int data sent by the core device a certain way. Either way, I'd say it's definitely worth a small amount of complications.

cjbe, 2021-06-11:
This would be such a quality of life improvement! 👍

dnadlinger, 2021-07-16:
A step on the way there would be a standalone tool (e.g. part of artiq_client) that resolves RTIO channel indices. This would still require adding some metadata to the device_db to resolve the channel ids, but would avoid messing with the EH machinery.