m-labs/artiq #1698: Frastino.set_dac() cannot get a float parameter

BallBall11, 2021-06-12:
# Bug Report

## One-Line Summary

When trying to set an output in float via Frastino.set_dac，an error occur.

## Issue Details

### Steps to Reproduce

1.  artiq-env.nix  is as below
```
  pkgs.mkShell {
    buildInputs = [
      (pkgs.python3.withPackages(ps: [
        # List desired Python packages here.

        # You probably want these two.
        artiq-full.artiq 
        artiq-full.artiq-comtools

        # You need a board support package if and only if you intend to flash
        # a board (those packages contain only board firmware).
        # The lines below are only examples, you need to select appropriate
        # packages for your boards.
        #artiq-full.artiq-board-kc705-nist_clock
        #artiq-full.artiq-board-kasli-wipm
        #ps.paramiko  # needed if and only if flashing boards remotely (artiq_flash -H)

        # The NixOS package collection contains many other packages that you may find
        # interesting for your research. Here are some examples:
        ps.pandas
        ps.numpy
        ps.scipy
        ps.jupyter
        ps.ipython
        ps.numba
        (ps.matplotlib.override { enableQt = true; })
        ps.bokeh
        #ps.cirq
        #ps.qiskit
      ]))

      # List desired non-Python packages here
      artiq-full.openocd  # needed if and only if flashing boards
      # Other potentially interesting packages from the NixOS package collection:
      #pkgs.gtkwave
      #pkgs.spyder
      #pkgs.R
      #pkgs.julia
      pkgs.qt5.full
      pkgs.qtcreator
    ];
  }
```

2. The following experiment:  
 ```
from artiq.experiment import  delay, us, s, ms
from artiq.language import units
from artiq.language.environment import EnvExperiment
from artiq.language.core import  kernel

class DAC(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.dac = self.get_device("fastino0")

    @kernel
    def run(self):
        self.core.reset()
        self.core.break_realtime()
        self.dac.init()
        delay(100*us)
        self.dac.set_dac(0,0)
        delay(100*us)
        self.dac.set_dac(0,5.5)
        
 ```

### Expected Behavior

Set the output at 5.5 Volt

### Actual (undesired) Behavior

* A type error occur
* `error: cannot unify float with numpy.int?  
        self.dac.set_dac(0,5.5)`

### Your System (omit irrelevant parts)

* Operating System: Linux/ nix
* ARTIQ version: ARTIQ v7.7639.9dee8bb9.beta
* Version of the gateware and runtime loaded in the core device: gateware ident 6.7604.040aa6fd
* Hardware involved:Kasli 2.0,  fastino 32x 16Bit DAC, IDC-SMA Adapter

pathfinder49, 2021-06-12:
I believe the artiq compiler is infering the argument to be an integer as you call `self.dac.set_dac(0,0)` before `self.dac.set_dac(0, 5.5)`. The artiq type inference can be quite picky. Your first usage of `set_dac` passes an integer as the voltage. This makes the compiler infer `set_dac()` as having 2 interger arguments leading to the error when called with a float.

BallBall11, 2021-06-13:
Thank you for your advice. That just solved my problem.

closed 2021-06-13 by BallBall11