m-labs/artiq #1702: applets.plot_hist: raise Exception on incorrect data lengths

hartytp, 2021-06-24:
Current behavior: if this test fails the applet silently ignores all data

https://github.com/m-labs/artiq/blob/094a346974642150f9df4d89197207d711e12088/artiq/applets/plot_hist.py#L26

Desired behavior: a clear exception (`ValueError`?) so users have a clue as to why their plot isn't showing anything.

sbourdeauducq, 2021-07-05:
> Desired behavior: a clear exception

Are you sure?
Unhandled exception will terminate the applet and the corresponding dock will go blank until the applet is restarted.
With asyncio it's even worse, the corresponding task will freeze and the error message will be displayed as a "task exception not retrieved" only when the applet is terminated from other causes.

I propose printing a textual warning on the plot instead. I will not suggest that you send a patch since you ought to focus on WRPLL.

hartytp, 2021-07-05:
> Are you sure?
> Unhandled exception will terminate the applet and the corresponding dock will go blank until the applet is restarted.

You're right, an exception probably isn't an appropriate solution here. Silently ignoring all data that doesn't have the right shape is annoying (particularly when there is no easily available documentation for the applet to set expectations straight). The issue I had was that our x-axis was bin centres rather than edges (so one fewer point) and it took me a little time to RTFS to figure out what was actually needed.

> Doesn't this happen transiently when new points are appended? 

I don't think so for this kind of histogram plot. At least for the workflow I had, we start with an x axis and (empty) histogram data. As points come in, we just increase the histogram counts, rather than changing the shape of any vector.

A text-based warning works well for me. Adding a message to the plot is potentially fine as well (although more work than adding 1 line for a logged warning).

dhslichter, 2021-07-06:
Agreed that an exception is not the way to go here.  Usually if one is plotting a histogram the arrays are bin edges and contents (see e.g. return values of `numpy.histogram` or `matplotlib.pyplot.histogram`), so the behavior of the applet as it currently stands seems to be fine to me.  Perhaps a little bit of documentation for the applets?  I don't see documentation there for any of these "basic" applets.

sbourdeauducq, 2021-07-07:
> > Doesn't this happen transiently when new points are appended?
> 
> I don't think so for this kind of histogram plot

@hartytp That's naive. X and Y are coming from two independent datasets and nothing guarantees that they are updated at the same time. In fact, they are *never* updated at the same time since the ``sync_struct`` messages are processed by a singlethread in the receiver.

So, yes, it happens transiently, and the warning should only be printed after the length mismatch has persisted for some amount of time.

How is WRPLL?

hartytp, 2021-07-07:
> @hartytp That's naive. X and Y are coming from two independent datasets and nothing guarantees that they are updated at the same time. In fact, they are never updated at the same time since the sync_struct messages are processed by a single thread in the receiver.

Yes, I realized that after my previous post but hadn't had a chance to update.

> So, yes, it happens transiently, and the warning should only be printed after the length mismatch has persisted for some amount of time.

That's a possibility, but given the length mismatch is not necessarily an indication of anything wrong I'm less inclined to add an error.

> Agreed that an exception is not the way to go here. Usually if one is plotting a histogram the arrays are bin edges and contents (see e.g. return values of numpy.histogram or [...]

This is the view I've come to. A docstring here (ideally accessible from the manual) which makes it clear what `x` and `y` are (and hence the length requirements) would resolve this well for me.

hartytp, 2021-07-07:
> How is WRPLL?

Unrelated to this issue. Happy to discuss, but this isn't the place.

sbourdeauducq, 2021-07-14:
Similar things should be done for plot_xy and plot_xy_hist.

closed 2021-07-19 by sbourdeauducq

hartytp, 2021-07-19:
Thanks! Other than the git stuff, the other small missing feature from the dashboard is `scan-devices`

PR #1718 (2021-07-14): applets: add warning message on plot for plot_hist

# ARTIQ Pull Request

## Description of Changes
Display a warning message on the plot window if the length of dataset X is not 1 longer than that of Y after a timeout of 0.5s. However, if the change in the dataset is less than the timeout, it will not display anything unless we change the dataset again. The display is as shown below:
[image]

### Related Issue
Closes #1702 

## Type of Changes

|   | Type |
| ------------- | ------------- |
| ✓  | :bug: Bug fix  |

## Steps (Choose relevant, delete irrelevant before submitting)

### All Pull  [... cut]
PR #1725 (2021-07-19): applets: add warning message on plot for `plot_xy_hist and fix bug`

# ARTIQ Pull Request

## Description of Changes
For the `plot_xy_hist` applet, the length of bin boundaries should be the same as the first dimension of histogram counts, and the length of point abscissas should be the same as the second dimension of histogram counts. Either failure will cause the xy plot to fail to display. The warning message is as shown below:
[image]
Potential bug fix: previously the histogram plot will crash in all my testing cases. Now, it will show `click on the data point at left to see histogram` if  [... cut]