m-labs/artiq #1706: [RFC] more git integration in dashboard

hartytp, 2021-06-25:
Maybe there are good reasons not to do this (merge conflicts etc) but I was wondering if it would make sense to add a bit more git integration into the dashboard. e.g. minimally, it would be nice to have an option to pull + scan instead of having to manually `git pull` then scan from the dashboard.

closed 2021-07-15 by hartytp

sbourdeauducq, 2021-07-15:
The master is using a bare git repository. It is feasible to add an option to the dashboard that makes the master pull some git URL into the bare repos, and bails out if it's not a fast-forward.

hartytp, 2021-07-15:
Yes, that would help on this specific thing. The reason I closed the issue is that there are larger issues that make the git integration in ARTIQ unergonomic at present. This is related to https://github.com/m-labs/artiq/issues/1543 etc, but also the fact that e.g. there isn't a convenient way of having multiple users running experiments from different branches.