m-labs/artiq #1707: Dashboard: create dataset

hartytp, 2021-06-28:
# ARTIQ Feature Request

## Problem this request addresses

Currently it's easy to edit / delete a dataset from the dashboard but not to create a new one. Is there a reason for this, or just an omission?

## Describe the solution you'd like

Option to create a new dataset from the dashboard.

hartytp, 2021-06-29:
The non-trivial issue here is how we deal with types. There isn't an obviously good solution (i.e. not `eval`ing user input) here that handles all cases in a clean way.

I had forgotten that one can create datasets using `artiq_client`. Maybe the best solution to this is to add a FAQ entry to explain how to create datasets via `artiq_client` (and why that's an appropriate solution)

dhslichter, 2021-06-29:
this is also something that has bugged me in the past.  I think I am like many other users in that pulling up a command line tool to do something like this doesn't feel very natural/smooth.  Is there no way that we can basically make a little GUI window that can pop up and lets you create a new dataset from the dashboard?  It could have a pull-down menu for dataset type, and then a place to type in the desired value.

hartytp, 2021-06-29:
> It would be OK with me if this only allows for, say, int32, int64, float, string, boolean, and numpy arrays or python lists of the above.

That's one option which I did consider, although once one starts thinking about e.g. multi-dimensional arrays it gets to be quite a wide range of types to support.

Now I've started to incorporate `artiq_client` in my workflow this feels like a non issue to me.

hartytp, 2021-06-29:
> Now I've started to incorporate artiq_client in my workflow this feels like a non issue to me.

Sorry, that came out a bit dismissive! Not intentional. Really all I meant was that this is messier to resolve than I'd initially thought and the command line tool is pretty nice to use once one gets used to it (and I've tended to have it in the background for a variety of other jobs like rescanning the device db)

dhslichter, 2021-06-29:
> That's one option which I did consider, although once one starts thinking about e.g. multi-dimensional arrays it gets to be quite a wide range of types to support.

Yeah, so have it just be 1D arrays, that would be fine.  I feel like multi-dimensional arrays are going to be things you enter programmatically anyway, and not manually.  

>Now I've started to incorporate artiq_client in my workflow this feels like a non issue to me.

From my perspective, though, it's very frustrating to use software that has a GUI if one cannot do the desired things from the GUI and has to open the command line interface instead.  I am thinking very much here of people getting started with ARTIQ, rather than experienced superusers.  The ability to just create a simple dataset from the GUI is quite a fundamental feature, and its absence is bewildering for new users -- I have seen this in person on multiple occasions.

dhslichter, 2021-06-29:
> the command line tool is pretty nice to use once one gets used to it (and I've tended to have it in the background for a variety of other jobs like rescanning the device db)

Agreed for experienced users, but I think that the ability to do simple things (e.g. make a dataset consisting of a single float) from the GUI is both good for lowering barriers for new users and would probably end up being used by experienced users for quick-n-dirty stuff too.  It seems like if one makes a little GUI functionality that just allows simple datasets to be set (and says "for other types, use the `artiq_client` tool or an experiment"), it would be simple, reasonably-scoped, and useful.

PR #1716 (2021-07-13): GUI: create new datasets

# ARTIQ Pull Request

## Description of Changes
Adding new datasets function in artiq_dashboard. The users can type in the name and value of the datasets in corresponding boxes. The data type will be shown next to the value box in real-time. It will show "N/A" if the input cannot be recognized and if the user still hits enter it will not proceed but show a log error. The decoder is `sipyco::pyon`
Warn!: string should be input like `"something"` with a double-quote.

[image]

### Related Issue

Closes #1707 

## Type of Changes

|   | Type |
| ------------- | --------- [... cut]
closed 2021-07-13 by sbourdeauducq